thopiekar/Cura-packaging #2: Remove dependency on dh-python

onitake, 2016-09-23:
I am by no means an expert, but I noticed that dpkg would generate (empty) python- packages alongside the python3- ones when dh-python is among the build dependencies. That doesn't make a lot of sense, as Python 2 is not supported by Uranium and libArcus.

Apparently, it suffices to simply depend on python3. The build system is cmake anyway.

onitake, 2016-09-29:
It looks like this dependency is required.
If the build system and other dependencies are specified correctly, debhelper will not try to build python2 packages.

thopiekar, 2016-09-29:
Nope, you are actually right. By passing ```--buildsystem=cmake``` you override the usage of `dh_python`.
I'm sure dh_python is superfluous here, but I'm currently to busy to check that.

onitake, 2016-09-29:
Interesting... Though, at it least it does no harm, if buildsystem is specified.
I'll leave it aside for now, it is a minor issue without any user impact.

commit 0f6434a3 (2016-10-19): master: Removing dependency on dh-python

Fixes #2
closed 2016-10-19 by thopiekar