thopiekar/Cura-packaging #3: License missing

onitake, 2016-09-29:
What is the license covering the packaging scripts?
Please add a global LICENSE file or specify it in each debian/copyright.
Thanks!

thopiekar, 2016-09-29:
Guess what, I never took time to choose or decide for one 😅 
So to start the discussion: Which one would you choose? And why?

onitake, 2016-09-29:
Hmm... It's not a major issue, since packaging script usually don't require that much of a creative effort.
But it's still important to be clear, or problems might arise later on.

For most of my projects, I prefer to use the 2-clause BSD license, because it leaves more freedom to the user than, say, the GPL. Cura is released under Affero GPL-3+, so adopting that license would make sense, too.
But I'm fine with anything that's compatible with Cura itself, so almost all the BSD licenses, GPL-3 and LGPL would work. LGPL would not make a lot of sense, though. :)

onitake, 2016-09-29:
These are the ones "preferred" by Debian: https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/
And here's a more complete list: https://spdx.org/licenses/

onitake, 2016-10-02:
I took the liberty of choosing the BSD 3-clause license for now, because it is very 'libre' and was already being used by some 3rd party sources in the Cura projects. This is just preliminary, I'm going to choose whatever you agree to ultimately.

I'm pretty much done with collecting all the copyright and licenses in each source tree, so I think it's time to finish this up and get it released. I'll also prepare a PR towards your repo once I'm done.

My recommended choices are:
- BSD-2-clause: Weak copyleft, gives the most freedom to users while disclaiming warranty. I like this license because of its simplicity. Also allows free use in commercial products.
- BSD-3-clause: Mostly the same as 2-clause, but does not allow advertising a derived product with the copyright holders' names unless explicitly permitted. Is being used by some 3rd party dependencies in the Cura projects, so would be easier to apply, as no extra license terms required in debian/copyright.
- AGPL3+: Same as the main source body of Cura. Easiest to apply, because no separate license terms would be required. Imposes some restrictions on use, but those don't have much of an impact on packaging. It would only cause problems if, say, Ultimaker started releasing packages under a non-free license while building them using your packaging scripts. I think this is very unlikely, and it would be only fair that they had to roll their own or ask your consent to relicensing first.

thopiekar, 2016-10-02:
Choosing BSD-3 sounds good. Apparently that's what Ultimaker does (done before they gave the possibility to fix problems for them). They mention my PPA in their FAQ without any credits.
So do I get it right? Choosing BSD-3 would mean that they have to add the credits?

onitake, 2016-10-02:
I'm not a lawyer, but I think giving credit (or not) only applies when you actually prepare a derivative work or a distribution, not if you just mention it on some website.
Also, the advertising clause actually does the opposite: Promoting a product using the original author's name is not allowed. I believe this clause was added originally so people could not advertise as if their derivative product was endorsed or supported by Berkeley University. It's of lesser importance when there's only a few authors.

IMHO, giving credit is always a sensible choice.
Can you point me to their FAQ so I can take a look?

thopiekar, 2016-10-02:
https://ultimaker.com/en/resources/21362-cura-2-1-linux-not-working

onitake, 2016-10-02:
I think you should just contact them and ask them to mention your name, if you like the promotion.
But since they point to your PPA directly, people would see it anyway. Your choice, I guess...
If they just copied everything and made it look like they were the authors/maintainers, it would be a bit different.

thopiekar, 2016-10-02:
Already contacted them months ago. It angered me a lot, because giving credits it the true minimum. However, now I see that they simply don't care (or have other priorities).
Don't want to put more blame on that, because I'm actually working with them together 😉
But having the correct licence would be great, so I could argue that they have to care more.

onitake, 2016-10-02:
Well, I think it's more important that Linux distributions carry proper packages by default. Users will certainly appreciate it when they can just `apt install cura` or `dnf install cura` or whatever applies for them.

I know that the Linux user base is still negligible compared to all those other computer users out there, but having an easy and comfortable way to install "standard" software is a big point in favour.

App stores are certainly better than having no package manager at all, but Linux or BSD distributions are clearly superior to anything else out there when it comes to update management.

thopiekar, 2022-06-21:
We are in 2022 now - moved to flatpaks now and will focus working on it.
Thus closing this here.

https://thopiekar.eu/2022/06/21/cura-rolled-out-as-flatpak/