thopiekar/Cura-packaging #4: cura 2.3

dmstan, 2016-10-22:
tried installing cura 2.3 using the ppa, but gives me the folowing error:
An uncaught exception has occurred!
Traceback (most recent call last):
  File "/usr/bin/cura", line 47, in <module>
    import Arcus #@UnusedImport
RuntimeError: the sip module implements API v11.0 to v11.2 but the Arcus module requires API v11.3
Any idea how to solve it?
Running Linux Mint 18 Cinnamon 64-bit
Thank you

thopiekar, 2016-10-23:
Hey @dmstan, looks for me like you've mixed up your installation.
The table here (https://www.linuxmint.com/download_all.php) says that Mint 18 is equivalent to Ubuntu Xenial. Is APT using Xenial, too?
Another possible reason could be that the Mint developers replaced the SIP with a newer version in their software repository.

dmstan, 2016-10-24:
I tried previously to install Cura-2.1.2-Linux.deb, version 2.1.3 deb and 2.3 deb from Ultimaker web-site. Maybe that mixed up stuff. Not sure how to fix.

thopiekar, 2016-10-24:
You'll need to remove their package first. You can take synaptic for this. Reitstall cura again and remove your config to be sure.

thopiekar, 2016-10-26:
I guess you managed to fix it.

closed 2016-10-26 by thopiekar