thopiekar/Cura-packaging #8: RuntimeError: the sip module implements API v11.0 to v11.2 but the Arcus module requires API v11.3

peteruithoven, 2016-12-15:
Can't boot a freshly installed version from the ppa, when booting through the commandline it gives the following error: 
```
$ cura
An uncaught exception has occurred!
Traceback (most recent call last):
  File "/usr/bin/cura", line 47, in <module>
    import Arcus #@UnusedImport
RuntimeError: the sip module implements API v11.0 to v11.2 but the Arcus module requires API v11.3
```

Installation process: 
I've purged my computer from both cura and cura-master ppa's packages:
```
$ sudo ppa-purge ppa:thopiekar/cura-master
Updating packages lists
PPA to be removed: thopiekar cura-master
Warning:  Could not find package list for PPA: thopiekar cura-master
$ sudo ppa-purge ppa:thopiekar/cura
Updating packages lists
PPA to be removed: thopiekar cura
Warning:  Could not find package list for PPA: thopiekar cura
```
Then reinstalled Cura from the Cura (stable) ppa:
```
$ sudo add-apt-repository ppa:thopiekar/cura
...
$ sudo apt update
...
$ sudo apt-get install cura
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libarcus libprotobuf11 python3-arcus python3-colorlog python3-decorator python3-numpy python3-protobuf
  python3-scipy python3-serial python3-uranium qml-module-qt-labs-folderlistmodel qml-module-qt-labs-settings
  qml-module-qtqml-models2 qml-module-qtquick-controls qml-module-qtquick-dialogs qml-module-qtquick-layouts
  qml-module-qtquick-privatewidgets qml-module-qtquick-window2 qml-module-qtquick2 uranium-plugins uranium-tools
Suggested packages:
  gfortran python-numpy-doc python3-dev python3-nose python3-numpy-dbg python-scipy-doc python3-wxgtk3.0
  | python3-wxgtk
The following NEW packages will be installed:
  cura libarcus libprotobuf11 python3-arcus python3-colorlog python3-decorator python3-numpy python3-protobuf
  python3-scipy python3-serial python3-uranium qml-module-qt-labs-folderlistmodel qml-module-qt-labs-settings
  qml-module-qtqml-models2 qml-module-qtquick-controls qml-module-qtquick-dialogs qml-module-qtquick-layouts
  qml-module-qtquick-privatewidgets qml-module-qtquick-window2 qml-module-qtquick2 uranium-plugins uranium-tools
0 upgraded, 22 newly installed, 0 to remove and 20 not upgraded.
Need to get 12,3 MB/17,9 MB of archives.
After this operation, 89,8 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qtquick-layouts amd64 5.5.1-1ubuntu1 [30,1 kB]
Get:2 http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial/main amd64 uranium-plugins all 1:2.3.99-truely-2.3.1-0~201612072003~rev1772~pkg113~ubuntu16.04.1 [24,8 kB]
Get:3 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qtquick-window2 amd64 5.5.1-2ubuntu6 [16,4 kB]
Get:4 http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial/main amd64 uranium-tools all 1:2.3.99-truely-2.3.1-0~201612072003~rev1772~pkg113~ubuntu16.04.1 [9790 B]
Get:5 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qtquick2 amd64 5.5.1-2ubuntu6 [35,8 kB]
Get:6 http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial/main amd64 libprotobuf11 amd64 3.1.0-0~201610252320~rev15~pkg76~ubuntu16.04.1 [671 kB]
Get:7 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qtquick-controls amd64 5.5.1-1ubuntu1 [643 kB]
Get:8 http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial/main amd64 libarcus amd64 1:2.3.99-truely-2.3.1-0~201612072003~rev112~pkg113~ubuntu16.04.1 [26,8 kB]
Get:9 http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial/main amd64 python3-arcus amd64 1:2.3.99-truely-2.3.1-0~201612072003~rev112~pkg113~ubuntu16.04.1 [17,6 kB]
Get:10 http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial/main amd64 python3-protobuf amd64 3.1.0-0~201610252320~rev15~pkg76~ubuntu16.04.1 [266 kB]
Get:11 http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial/main amd64 python3-uranium all 1:2.3.99-truely-2.3.1-0~201612072003~rev1772~pkg113~ubuntu16.04.1 [205 kB]
Get:12 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 python3-numpy amd64 1:1.11.0-1ubuntu1 [1762 kB]
Get:13 http://nl.archive.ubuntu.com/ubuntu xenial/universe amd64 python3-decorator all 4.0.6-1 [9388 B]
Get:14 http://nl.archive.ubuntu.com/ubuntu xenial/universe amd64 python3-scipy amd64 0.17.0-1 [8327 kB]
Get:15 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 python3-serial all 3.0.1-1 [54,1 kB]                     
Get:16 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qtqml-models2 amd64 5.5.1-2ubuntu6 [16,8 kB]  
Get:17 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qtquick-privatewidgets amd64 5.5.1-1ubuntu1 [38,9 kB]
Get:18 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qtquick-dialogs amd64 5.5.1-1ubuntu1 [89,0 kB]
Get:19 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qt-labs-folderlistmodel amd64 5.5.1-2ubuntu6 [30,9 kB]
Get:20 http://nl.archive.ubuntu.com/ubuntu xenial/main amd64 qml-module-qt-labs-settings amd64 5.5.1-2ubuntu6 [20,1 kB]
Get:21 http://nl.archive.ubuntu.com/ubuntu xenial/universe amd64 python3-colorlog all 2.6.0-1 [7912 B]                
[... 75 lines cut ...]
Processing triggers for shared-mime-info (1.5-2ubuntu0.1) ...
Setting up qml-module-qtquick-layouts:amd64 (5.5.1-1ubuntu1) ...
Setting up qml-module-qtquick-window2:amd64 (5.5.1-2ubuntu6) ...
Setting up qml-module-qtquick2:amd64 (5.5.1-2ubuntu6) ...
Setting up qml-module-qtquick-controls:amd64 (5.5.1-1ubuntu1) ...
Setting up uranium-plugins (1:2.3.99-truely-2.3.1-0~201612072003~rev1772~pkg113~ubuntu16.04.1) ...
Setting up uranium-tools (1:2.3.99-truely-2.3.1-0~201612072003~rev1772~pkg113~ubuntu16.04.1) ...
Setting up libprotobuf11:amd64 (3.1.0-0~201610252320~rev15~pkg76~ubuntu16.04.1) ...
Setting up libarcus:amd64 (1:2.3.99-truely-2.3.1-0~201612072003~rev112~pkg113~ubuntu16.04.1) ...
Setting up python3-arcus (1:2.3.99-truely-2.3.1-0~201612072003~rev112~pkg113~ubuntu16.04.1) ...
Setting up python3-numpy (1:1.11.0-1ubuntu1) ...
Setting up python3-decorator (4.0.6-1) ...
Setting up python3-scipy (0.17.0-1) ...
Setting up python3-protobuf (3.1.0-0~201610252320~rev15~pkg76~ubuntu16.04.1) ...
Setting up python3-serial (3.0.1-1) ...
Setting up qml-module-qtqml-models2:amd64 (5.5.1-2ubuntu6) ...
Setting up qml-module-qtquick-privatewidgets:amd64 (5.5.1-1ubuntu1) ...
Setting up qml-module-qtquick-dialogs:amd64 (5.5.1-1ubuntu1) ...
Setting up python3-uranium (1:2.3.99-truely-2.3.1-0~201612072003~rev1772~pkg113~ubuntu16.04.1) ...
Setting up qml-module-qt-labs-folderlistmodel:amd64 (5.5.1-2ubuntu6) ...
Setting up qml-module-qt-labs-settings:amd64 (5.5.1-2ubuntu6) ...
Setting up cura (1:2.3.99-truely-2.3.1-0~201612120102~rev1403~pkg113~ubuntu16.04.1) ...
Setting up python3-colorlog (2.6.0-1) ...
Processing triggers for libc-bin (2.23-0ubuntu5) ...
```

peteruithoven, 2016-12-16:
With another `upgrade` this problem seems fixed.

peteruithoven, 2016-12-19:
After purging the stable ppa, re-adding it, installing cura and cura-extra-plugins-all and booting Cura from the commandline I got this error again.

When running apt upgrade there where quite some updates:
```
The following packages will be upgraded:
  cura-engine python-sip python3-pyqt5 python3-pyqt5.qtopengl python3-pyqt5.qtquick python3-pyqt5.qtsvg python3-sip
```
After the upgrade Cura started just fine. Could it be that there are some dependencies not properly defined?

thopiekar, 2016-12-19:
Hey @peteruithoven ,
I sadly didn't get any notification about your issue report in my email.

You should use `sudo apt dist-upgrade` instead of `sudo apt dist-upgrade`, because `upgrade` only updates packages unless there is no need to install new packages. Purging the PPA shouldn't be needed.

Hmm, I'll take out my notebook (where I only use official releases on Windows and my stable PPA on Ubuntu). I kicked out some dependencies (QML), which were set before for 2.1, but I'm almost sure they are not needed anymore.

thopiekar, 2016-12-20:
Ok, resolved a conflict here, while cura-resources-materials gets unpacked.
After resolving this problem starting cura works very well (at least for me).

peteruithoven, 2016-12-20:
No problem at all. 
> You should use `sudo apt dist-upgrade` instead of `sudo apt dist-upgrade`

You mean `sudo apt dist-upgrade` instead of `sudo apt upgrade`? 

What kind of conflict? 

I tried running the following again and got the same error. 
- `sudo ppa-purge ppa:thopiekar/cura`
- `sudo add-apt-repository ppa:thopiekar/cura`
- `sudo apt update`
- `sudo apt install cura`

Wouldn't people installing the ppa for the first time get the same issue? Or is that just my linux package managment noobness talking?

thopiekar, 2016-12-20:
Correct. See http://askubuntu.com/questions/81585/what-is-dist-upgrade-and-why-does-it-upgrade-more-than-upgrade

Hmm, I don't think so. People who are installing cura for the first time might have also the benefit that all plugins get installed (note that I splitted out some plugins from Cura, eg like sliceinfo for those who care a lot about privacy).
Something is strange on your setup. If you've got teamviewer installed I can take a look on it, of course :wink:

peteruithoven, 2016-12-20:
> People who are installing cura for the first time might have also the benefit that all plugins get installed (note that I splitted out some plugins from Cura, eg like sliceinfo for those who care a lot about privacy).

Wait, so you split out plugins, but people get the benefit that all plugins get installed? That seems counter intuitive. 

> Something is strange on your setup. If you've got teamviewer installed I can take a look on it, of course :wink:

Not sure how I feel about giving people that kind of access. Maybe we just see if other people run into the same issue. Maybe it's caused by my usage of ppa-purge somehow.

thopiekar, 2016-12-20:
Well, this is how you think about it, but there are also people, who only want essential features of Cura or would like to have SliceInfo (the plugin which sends stats to Ultimaker) removed.

Ok, but then let's find the reason for your upgrade problem here.
Yes, that's what I wanted to check. My output for example looks like this:
```
thopiekar@home:~/Desktop/Zwischenspeicher$ sudo ppa-purge ppa:thopiekar/cura
[sudo] Passwort für thopiekar: 
Updating packages lists
PPA to be removed: thopiekar cura
Warning:  Could not find package list for PPA: thopiekar cura
thopiekar@home:~/Desktop/Zwischenspeicher$ sudo ppa-purge ppa:thopiekar/cura-master
Updating packages lists
PPA to be removed: thopiekar cura-master
Package revert list generated:
 cura- cura-engine- cura-extra-plugin-octoprint- cura-extra-plugin-orientation- 
cura-extra-plugin-postprocessing- cura-extra-plugins-all- 
cura-plugin-sliceinfo- cura-plugin-ultimakermachineactions- 
cura-plugin-um3networkprinting- cura-plugin-x3dreader- cura-resources-firmware- 
cura-resources-materials- libarcus:amd64- libarcus-dev:amd64- 
libprotobuf-dev:amd64- libprotobuf-lite11:amd64- libprotobuf11:amd64- 
libprotoc-dev:amd64- libprotoc11:amd64- protobuf-compiler- python3-arcus- 
python3-uranium- uranium-plugins-

Disabling thopiekar PPA from 
/etc/apt/sources.list.d/thopiekar-ubuntu-cura-wily.list
Disabling thopiekar PPA from 
/etc/apt/sources.list.d/thopiekar-ubuntu-cura-master-yakkety.list
Updating packages lists
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  libatk-bridge2.0-dev libatspi2.0-dev libepoxy-dev libibus-1.0-dev
  libinput-dev libmircookie-dev libmircookie2 libpng12-0:i386
  libqt5location5-plugins libqt5positioning5-plugins libqt5scripttools5
  libsndio-dev libwayland-bin libwayland-dev libxkbcommon-dev python3-colorlog
  python3-netifaces python3-zeroconf qml-module-qtqml-models2
  wayland-protocols
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
Die folgenden Pakete werden ENTFERNT:
  cura cura-engine cura-extra-plugin-octoprint cura-extra-plugin-orientation
  cura-extra-plugin-postprocessing cura-extra-plugins-all
  cura-plugin-sliceinfo cura-plugin-ultimakermachineactions
  cura-plugin-um3networkprinting cura-plugin-x3dreader cura-resources-firmware
  cura-resources-materials libarcus libarcus-dev libegl1-mesa-dev
  libgles2-mesa-dev libgtk-3-dev libmirclient-dev libmircommon-dev
  libprotobuf-dev libprotobuf-lite11 libprotobuf11 libprotoc-dev libprotoc11
  libqt5opengl5-dev libqt5sensors5-dev libqt5serialport5-dev libqt5webkit5-dev
  libqt5websockets5-dev libqt5x11extras5-dev libqt5xmlpatterns5-dev
  libsdl2-dev libsdl2-image-dev libsdl2-mixer-dev libsdl2-ttf-dev
  protobuf-compiler python3-arcus python3-uranium qtbase5-dev
  qtbase5-private-dev qtdeclarative5-dev qtdeclarative5-private-dev
  qtlocation5-dev qtmultimedia5-dev qtpositioning5-dev qtscript5-dev
  qtscript5-private-dev qttools5-dev qttools5-private-dev uranium-plugins
0 aktualisiert, 0 neu installiert, 50 zu entfernen und 16 nicht aktualisiert.
Nach dieser Operation werden 111 MB Plattenplatz freigegeben.
N: Datei »50unattended-upgrades.ucf-dist« in Verzeichnis »/etc/apt/apt.conf.d/« wird ignoriert, da sie eine ungültige Dateinamen-Erweiterung hat.
Möchten Sie fortfahren? [J/n]
```
In short: I'm asked whether I want to remove the old packages, but I don't see this question on your side.

thopiekar, 2016-12-20:
Btw. all plugins are still installed unless Apt is configured to install recommendations.
So if you install cura, all plugins are going to be installed, but using a package manager (GUI) you can unselect one of these before installing.

```
thopiekar@home:~/Desktop/Zwischenspeicher$ sudo apt install cura
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden zusätzlichen Pakete werden Installiert
  cura-engine cura-plugin-sliceinfo cura-plugin-ultimakermachineactions
  cura-plugin-um3networkprinting cura-plugin-x3dreader cura-resources-firmware
  cura-resources-materials libarcus libprotobuf11 python3-arcus
  python3-uranium uranium-plugins
Die folgenden NEUEN Pakete werden installiert:
  cura cura-engine cura-plugin-sliceinfo cura-plugin-ultimakermachineactions
  cura-plugin-um3networkprinting cura-plugin-x3dreader cura-resources-firmware
  cura-resources-materials libarcus libprotobuf11 python3-arcus
  python3-uranium uranium-plugins
0 aktualisiert, 13 neu installiert, 0 zu entfernen und 16 nicht aktualisiert.
Es müssen noch 6.780 kB von 7.665 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 45,2 MB Plattenplatz zusätzlich benutzt.
N: Datei »50unattended-upgrades.ucf-dist« in Verzeichnis »/etc/apt/apt.conf.d/« wird ignoriert, da sie eine ungültige Dateinamen-Erweiterung hat.
Möchten Sie fortfahren? [J/n]
```

peteruithoven, 2016-12-20:
> Btw. all plugins are still installed unless Apt is configured to install recommendations.

So plugins like `cura-plugin-um3networkprinting` are installed automatically because it's a recommendation, unless apt is configured not to? 

This is my ppa-purge output: 
```
$ sudo ppa-purge ppa:thopiekar/cura
[sudo] password for peteruithoven: 
Updating packages lists
PPA to be removed: thopiekar cura
Package revert list generated:
 cura- cura-engine/xenial cura-extra-plugin-octoprint- 
cura-extra-plugin-orientation- cura-extra-plugin-postprocessing- 
cura-extra-plugins-all- cura-plugin-sliceinfo- 
cura-plugin-ultimakermachineactions- cura-plugin-um3networkprinting- 
cura-plugin-x3dreader- cura-resources-firmware- cura-resources-materials- 
libarcus:amd64- libprotobuf11:amd64- python-sip/xenial python3-arcus- 
python3-pyqt5/xenial python3-pyqt5.qtopengl/xenial python3-pyqt5.qtquick/xenial 
python3-pyqt5.qtsvg/xenial python3-sip/xenial python3-uranium- uranium-plugins-

Disabling thopiekar PPA from 
/etc/apt/sources.list.d/thopiekar-ubuntu-cura-master-xenial.list
Disabling thopiekar PPA from 
/etc/apt/sources.list.d/thopiekar-ubuntu-cura-xenial.list
Updating packages lists
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Selected version '14.01-2' (Ubuntu:16.04/xenial [amd64]) for 'cura-engine'
Selected version '4.17+dfsg-1build1' (Ubuntu:16.04/xenial [amd64]) for 'python-sip'
Selected version '5.5.1+dfsg-3ubuntu4' (Ubuntu:16.04/xenial [amd64]) for 'python3-pyqt5'
Selected version '5.5.1+dfsg-3ubuntu4' (Ubuntu:16.04/xenial [amd64]) for 'python3-pyqt5.qtopengl'
Selected version '5.5.1+dfsg-3ubuntu4' (Ubuntu:16.04/xenial [amd64]) for 'python3-pyqt5.qtquick'
Selected version '5.5.1+dfsg-3ubuntu4' (Ubuntu:16.04/xenial [amd64]) for 'python3-pyqt5.qtsvg'
Selected version '4.17+dfsg-1build1' (Ubuntu:16.04/xenial [amd64]) for 'python3-sip'
The following packages were automatically installed and are no longer required:
  python3-colorlog python3-decorator python3-netifaces python3-numpy python3-scipy python3-serial python3-zeroconf
  qml-module-qtqml-models2 qml-module-qtquick-controls qml-module-qtquick-dialogs qml-module-qtquick-layouts
  qml-module-qtquick-privatewidgets qml-module-qtquick-window2 qml-module-qtquick2
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libpolyclipping16
Suggested packages:
  python3-pyqt5-dbg
The following packages will be REMOVED:
  cura cura-extra-plugin-octoprint cura-extra-plugin-orientation cura-extra-plugin-postprocessing
  cura-extra-plugins-all cura-plugin-sliceinfo cura-plugin-ultimakermachineactions cura-plugin-um3networkprinting
  cura-plugin-x3dreader cura-resources-firmware cura-resources-materials libarcus libprotobuf11 python3-arcus
  python3-uranium uranium-plugins
The following NEW packages will be installed:
  libpolyclipping16
The following packages will be DOWNGRADED:
  cura-engine python-sip python3-pyqt5 python3-pyqt5.qtopengl python3-pyqt5.qtquick python3-pyqt5.qtsvg python3-sip
0 upgraded, 1 newly installed, 7 downgraded, 16 to remove and 0 not upgraded.
Need to get 43,8 kB/2776 kB of archives.
After this operation, 46,3 MB disk space will be freed.
Do you want to continue? [Y/n] 
```

peteruithoven, 2016-12-20:
> Well, this is how you think about it, but there are also people, who only want essential features of Cura or would like to have SliceInfo (the plugin which sends stats to Ultimaker) removed.

I don't have a clear preference in regards to the pluginsreally, I'm just curious. My main concern has been getting Cura to update with the rest of my system.

thopiekar, 2016-12-20:
Yes, and using recommendations is default - at least on Ubuntu. You are using elementary os, isn't it?

The ppa-purge output looks fine. Don't see any reason why it shouldn't work.

When I read your first message here, I couldn't find any error messages. So as you where using ppa-purge I guessed you are doing it for a good reason.
Generally using your distribution's update manager or `apt dist-upgrade` from the console should be more then enough to update/upgrade Cura. `ppa-purge` is only needed if you switch between PPAs like stable and master.

peteruithoven, 2016-12-20:
> Yes, and using recommendations is default - at least on Ubuntu. You are using elementary os, isn't it?
Ah, but not all plugins? For example the post-processing plugin isn't? 

The ppa description still mentions the following btw, which seems wrong if Ultimaker 3 network-printing support is usually installed because it's recommended? 
> To install extra plugins, eg. for Ultimaker 3 network-printing support, Octoprint printing, etc., do:
> 
> $ sudo apt-get install cura-extra-plugins-all

I was indeed using ppa-purge to switch between the master and stable and then again to try to reproduce the issue I had. 

> When I read your first message here, I couldn't find any error messages
True, there is only an error when I start Cura

To reiterate, I'm confused that a `apt dist-upgrade` is necessary after installing cura to get it to run. (on my system at least)

thopiekar, 2016-12-20:
This info has not been updated to the new packaging layout so far.

Ok, then.

Since you are using Elementary OS, which is based on Xenial, you need to do a `dist-upgrade` because my PPA comes with a more recent version of _Protobuf_. Cura is based on libArcus and it is built on top of a newer version now. If you only do a "simple" upgrade using `apt upgrade` apt will refuse to update libarcus since a new package with a newer _Protobuf_ library is additionally needed.

peteruithoven, 2016-12-20:
Thanks for the explanation, maybe I'm spoiled with npm (Node.js) but isn't there a kind of dependency managment? Where Cura indicates it requires libArcus, which in turn indicates it needs a certain version of Protobuf? 

My main point point why would I need to run a dist-upgrade / upgrade at all? Shouldn't `apt install cura` take care of it's dependencies?

thopiekar, 2016-12-20:
Of course there is. That's what you get if you are doing install/upgrade/dist-upgrade actions. I added a virtual package called `cura-plugin-api-*` to the cura package. So you can't install older packages.
The difference between upgrade and dist-upgrade is only how "strong" an upgrade is, as you probably read from the link I sent you before.
Yes, `apt install cura` should almost do the same. But I don't use this way of updating packages and don't want to give any guarantee because there is already a command which is meant to be used for upgrading tasks, called `upgrade` or `dist-upgrade` (depends on what you like to do).

Simply to update your packages. You usually use `install` only if you are installing something for the first time. It is just the normal way to do updates..
```
APT(8)                                                                                                                         APT                                                                                                                        APT(8)

NAME
       apt - command-line interface

SYNOPSIS
       apt [-h] [-o=config_string] [-c=config_file] [-t=target_release] [-a=architecture] {list | search | show | update | install pkg [{=pkg_version_number | /target_release}]...  | remove pkg...  | upgrade | full-upgrade | edit-sources | {-v | --version}
           | {-h | --help}}

DESCRIPTION
       apt provides a high-level commandline interface for the package management system. It is intended as an end user interface and enables some options better suited for interactive usage by default compared to more specialized APT tools like apt-get(8)
       and apt-cache(8).

       Much like apt itself, its manpage is intended as an end user interface and as such only mentions the most used commands and options partly to not duplicate information in multiple places and partly to avoid overwhelming readers with a cornucopia of
       options and details.

       update (apt-get(8))
           update is used to download package information from all configured sources. Other commands operate on this data to e.g. perform package upgrades or search in and display details about all packages available for installation.

       upgrade (apt-get(8))
           upgrade is used to install available upgrades of all packages currently installed on the system from the sources configured via sources.list(5). New packages will be installed if required to statisfy dependencies, but existing packages will
           never be removed. If an upgrade for a package requires the remove of an installed package the upgrade for this package isn't performed.

       full-upgrade (apt-get(8))
           full-upgrade performs the function of upgrade but will remove currently installed packages if this is needed to upgrade the system as a whole.
```
(`dist-upgrade` is a synonym to `dist-upgrade`, which is used in the `apt-get` command)

closed 2016-12-23 by thopiekar