thopiekar/Cura-packaging #12: linux mint 18.1 (aka ubuntu 16.04) main view is black

jnweiger, 2016-12-28:
I've tried the official 2.4.0 beta and 2.4.0 from your ppa, in both cases the main object view remains black and flickers with background contents when moving the mouse. 

Also, there are error messages scrolling thourgh the console 
<pre>
2016-12-28 10:45:20,952 - CRITICAL - cura.CrashHandler.show [32]: An uncaught exception has occurred!
c Traceback (most recent call last):
2016-12-28 10:45:20,952 - CRITICAL - cura.CrashHandler.show [36]: Traceback (most recent call last):
c   File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/MainWindow.py", line 177, in _render
2016-12-28 10:45:20,952 - CRITICAL - cura.CrashHandler.show [36]:   File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/MainWindow.py", line 177, in _render
c     renderer.beginRendering()
2016-12-28 10:45:20,953 - CRITICAL - cura.CrashHandler.show [36]:     renderer.beginRendering()
c   File "/usr/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 115, in beginRendering
2016-12-28 10:45:20,953 - CRITICAL - cura.CrashHandler.show [36]:   File "/usr/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 115, in beginRendering
c     self._initialize()
2016-12-28 10:45:20,953 - CRITICAL - cura.CrashHandler.show [36]:     self._initialize()
c   File "/usr/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 169, in _initialize
2016-12-28 10:45:20,954 - CRITICAL - cura.CrashHandler.show [36]:   File "/usr/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 169, in _initialize
c     OpenGL.setInstance(QtOpenGL())
2016-12-28 10:45:20,954 - CRITICAL - cura.CrashHandler.show [36]:     OpenGL.setInstance(QtOpenGL())
c   File "/usr/lib/python3/dist-packages/UM/Qt/GL/QtOpenGL.py", line 62, in __init__
2016-12-28 10:45:20,954 - CRITICAL - cura.CrashHandler.show [36]:   File "/usr/lib/python3/dist-packages/UM/Qt/GL/QtOpenGL.py", line 62, in __init__
c     self._gpu_type = self._gl.glGetString(self._gl.GL_RENDERER)
2016-12-28 10:45:20,954 - CRITICAL - cura.CrashHandler.show [36]:     self._gpu_type = self._gl.glGetString(self._gl.GL_RENDERER)
c UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 21: ordinal not in range(128)
2016-12-28 10:45:20,955 - CRITICAL - cura.CrashHandler.show [36]: UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 21: ordinal not in range(128)
</pre>
[image]

any hints here?

thopiekar, 2016-12-28:
Hey @jnweiger,

yes, this is an known issue in Uranium. I just prepared a PR for it, as it became forgotten to be applied so far.

jnweiger, 2016-12-29:
This patch fixes everything. Thanks.

closed 2016-12-29 by jnweiger