thopiekar/Cura-packaging #13: Stick to 2.3 as stable (for now)

arthurlutz, 2017-01-19:
It seems that 2.4 is still labelled beta on https://ultimaker.com/en/products/cura-software

Maybe https://launchpad.net/%7Ethopiekar/+archive/ubuntu/cura should still host the 2.3 version ?

thopiekar, 2017-02-05:
Hmm, yes, I should probably keep the latest stable. I was sure Cura is stable enough, but after seeing that people had a lot of problems with materials and configuration migration that I felt bad about making this update.

However, maintaining a third PPA for betas is currently no option. I'm quite busy at the moment and if someone really wants a PPA for betas, then she/he should jump in and help out maintaining the PPAs.

thopiekar, 2017-03-02:
Using now the 2.4.0 tag, like Ultimaker uses for their official builds.
So closing this ticket therefore...

closed 2017-03-02 by thopiekar