thopiekar/Cura-packaging #14: Have a recommends depency on numpy-stl

arthurlutz, 2017-01-19:
Looking at the cura log at startup I get : 

```
2017-01-19 21:40:09,705 - WARNING - STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
```
So I took at look at https://github.com/WoLpH/numpy-stl/ which doesn't have a package or PPA (or I haven't found it)

I added an issue here : https://github.com/WoLpH/numpy-stl/issues/46

It would be awesome if your PPA's could host that recommended dependency. 

(btw thanks a bunch for your PPA... avoids having to fill in your info at ultimaker... and getting auto updates)

thopiekar, 2017-02-05:
No problem, @arthurlutz. Sorry for the late reply.

Yes, I was planning to package numpy-stl, but there were problems with it in Cura. It was affecting some STL files (in binary format??). I can't remember exactly. However, that is the reason why I stopped working on it...

Uvar, 2017-03-17:
@thopiekar 
I am not sure how it would generate problems as it concerns misspecified polygons, but some digging through the STLReader made it clear that it uses the default mesh load behaviour. This is to discard polygons which have 0 area according to the cross-product of the vectors. As the keyword argument is accessible, this default behaviour can be overridden.

Maybe this could fix the issue you found?

commit c1dc1589 (2017-04-29): 3rd-party: numpy-stl: Adding packaging files for Numpy-STL

However, a build recipe still needs to be created..

Fixes: #14
closed 2017-04-29 by thopiekar