thopiekar/Cura-packaging #16: When installing Cura the fdm_materials folder is not being installed

maukcc, 2017-03-13:
When installing a fresh Cura the fdm_materials folder is not being installed in /usr/share/cura/resources
Am I doing something wrong?

thopiekar, 2017-04-29:
Hmm, according to my packaging files it should be there, since ```cura``` depends on ```cura-resources-materials```

thopiekar, 2017-04-30:
Since cura depends on cura-resource-materials - see https://github.com/thopiekar/Cura-packaging/blob/master/Cura/control#L25 -  it really should be there when installing it.
Therefore...

closed 2017-04-30 by thopiekar

maukcc, 2017-05-01:
Sorry, that only happened for a couple of days for some reason, and works good now.

maukcc, 2017-05-03:
No, I tried it again and no materials folder is being generated in /usr/share/cura/resources/

I am using PPA: http://ppa.launchpad.net/thopiekar/cura-master/ubuntu xenial main
I purged cura, deleted the /usr/share/cura folder and it installed everything except the materials

When I then do:
sudo apt install cura-resources-materials
it tells me it is already the latest version

thopiekar, 2017-05-03:
Sorry, but this is really not possible: 
[image]
Also the last build log confirms, that the files are inside the package:
https://launchpadlibrarian.net/317964740/buildlog_ubuntu-xenial-amd64.fdm-materials_2.5.99-master~201705021147~rev156~pkg154~ubuntu16.04.1_BUILDING.txt.gz

maukcc, 2017-05-03:
I am using a 32bit laptop, will that be of influence?

maukcc, 2017-05-03:
Also, I believe there used to be a folder called firmware, and that is not there any more either.
When I purged cura, I had to manually delete these 2 folders, firmware and materials

maukcc, 2017-05-03:
It works.
when I manually deleted the 2 folders (firmware and materials) it put them in the rubbish bin, and for some reason it did not re-install them.
I deleted them from the rubbish bin and now it has installed them.