thopiekar/Cura-packaging #18: Install cura from PPA on Debian 9

userosos, 2017-04-29:
Hello!
I can not get cura from the PPA in my Debian 9 system. 

sudo add-apt-repository ppa:thopiekar/cura-master
```
 Cura built from master! Only for experienced users or developers!

Feedback/Questions/Used packaging files:
########################################

See: https://github.com/thopiekar/Cura-packaging
 More info: https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keybox '/tmp/tmpeakpmljd/pubring.gpg' created
gpg: /tmp/tmpeakpmljd/trustdb.gpg: trustdb created
gpg: key D51DB14E9FFECCF3: public key "Launchpad PPA for Thomas Pietrowski" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: no valid OpenPGP data found.
```
after i do apt update:

```
Чтение списков пакетов… Готово
E: Репозиторий «http://ppa.launchpad.net/thopiekar/cura-master/ubuntu artful Release» не содержит файла Release.
N: Обновление из этого репозитория нельзя выполнить безопасным способом, и поэтому по умолчанию он отключён.
N: Смотрите справочную страницу apt-secure(8) о создании репозитория и настройке пользователя.
```

thopiekar, 2017-04-29:
As far as I see from the second message it expects Ubuntu artful which is now the development release.
I wonder why Debian is choosing it. However, will enable builds for artful. I can't give any guarantees etc. you have to experiment on your own to get it fully working at the end..

userosos, 2017-04-29:
@thopiekar 
Thank you! I change in 
/etc/apt/sources.list.d/thopiekar-ubuntu-cura-artful.list
to
deb http://ppa.launchpad.net/thopiekar/cura/ubuntu zesty main
deb-src http://ppa.launchpad.net/thopiekar/cura/ubuntu zesty main
And it work for me in debian stretch 9.

thopiekar, 2017-04-29:
Nice! However, I would recommend to use the stable PPA if possible :wink: 

However, happy slicing!

closed 2017-04-29 by thopiekar