thopiekar/Cura-packaging #25: Cura crashes on Ubuntu 16.04

chrisspen, 2017-07-18:
Attempting to launch Cura on Ubuntu 16.04 returns the error:

    2017-07-18 19:07:39,679 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin ConsoleLogger
    2017-07-18 19:07:39,690 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
    2017-07-18 19:07:39,691 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin CuraEngineBackend
    2017-07-18 19:07:39,696 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin MirrorTool
    2017-07-18 19:07:39,703 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin ScaleTool
    2017-07-18 19:07:39,706 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin CameraTool
    2017-07-18 19:07:39,709 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin SelectionTool
    2017-07-18 19:07:39,714 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin TranslateTool
    2017-07-18 19:07:39,719 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin RotateTool
    2017-07-18 19:07:39,719 - WARNING - UM.PluginRegistry.loadPlugin [142]: Plugin ConsoleLogger was already loaded
    2017-07-18 19:07:39,725 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin LocalFileOutputDevice
    2017-07-18 19:07:39,727 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin FileLogger
    2017-07-18 19:07:39,729 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin OBJReader
    2017-07-18 19:07:39,732 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin STLWriter
    2017-07-18 19:07:39,734 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin OBJWriter
    2017-07-18 19:07:39,738 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
    2017-07-18 19:07:39,739 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin STLReader
    2017-07-18 19:07:39,742 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin SimpleView
    2017-07-18 19:07:39,748 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin GCodeReader
    2017-07-18 19:07:39,750 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin CuraProfileWriter
    2017-07-18 19:07:39,755 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin PostProcessingPlugin
    2017-07-18 19:07:39,767 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin X3DReader
    2017-07-18 19:07:39,768 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin GCodeWriter
    2017-07-18 19:07:39,770 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin LegacyProfileReader
    2017-07-18 19:07:39,773 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin 3MFWriter
    2017-07-18 19:07:39,776 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
    2017-07-18 19:07:39,776 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin PerObjectSettingsTool
    2017-07-18 19:07:39,780 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin LayerView
    2017-07-18 19:07:39,820 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin SliceInfoPlugin
    2017-07-18 19:07:39,837 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin OctoPrintPlugin
    2017-07-18 19:07:39,850 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!
    2017-07-18 19:07:39,851 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):
    2017-07-18 19:07:39,851 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/bin/cura", line 69, in <module>
    2017-07-18 19:07:39,852 - CRITICAL - cura.CrashHandler.show [42]:     app = cura.CuraApplication.CuraApplication.getInstance()
    2017-07-18 19:07:39,852 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/Application.py", line 316, in getInstance
    2017-07-18 19:07:39,852 - CRITICAL - cura.CrashHandler.show [42]:     Application._instance = cls()
    2017-07-18 19:07:39,852 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 195, in __init__
    2017-07-18 19:07:39,852 - CRITICAL - cura.CrashHandler.show [42]:     super().__init__(name = "cura", version = CuraVersion, buildtype = CuraBuildType)
    2017-07-18 19:07:39,853 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 110, in __init__
    2017-07-18 19:07:39,853 - CRITICAL - cura.CrashHandler.show [42]:     self._loadPlugins()
    2017-07-18 19:07:39,853 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 490, in _loadPlugins
    2017-07-18 19:07:39,853 - CRITICAL - cura.CrashHandler.show [42]:     self._plugin_registry.loadPlugins()
    2017-07-18 19:07:39,853 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 199, in loadPlugins
    2017-07-18 19:07:39,854 - CRITICAL - cura.CrashHandler.show [42]:     plugin_data = self.getMetaData(plugin_id)
    2017-07-18 19:07:39,854 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 219, in getMetaData
    2017-07-18 19:07:39,854 - CRITICAL - cura.CrashHandler.show [42]:     if not self._populateMetaData(plugin_id):
    2017-07-18 19:07:39,854 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 334, in _populateMetaData
    2017-07-18 19:07:39,854 - CRITICAL - cura.CrashHandler.show [42]:     raise InvalidMetaDataError(plugin_id)
    2017-07-18 19:07:39,855 - CRITICAL - cura.CrashHandler.show [42]: UM.PluginError.InvalidMetaDataError: Invalid metadata for plugin OrientationPlugin

thopiekar, 2017-08-05:
Looks like you are using Cura master, right?

chrisspen, 2017-08-05:
I'm using the packages from the PPA. Removing the cura plugins package fixes it, but now there are no plugins. That's fine for my use, but I'd imagine the plugins package should break Cura.

thopiekar, 2017-08-05:
One of the plugins has not been updated. There is a new API now, how to register plugins.
I just wonder why Cura is not just ignoring it instead of crashing.

thopiekar, 2017-08-05:
Additionally there are two PPAs. One with stable software and one with master (means development state)..

thopiekar, 2017-09-28:
@chrisspen Is it still the case? I had the issue in the past, too, and now it is gone.

closed 2017-11-07 by thopiekar