thopiekar/Cura-packaging #27: Cura (master) does not slice

mahowi, 2017-09-26:
Because Ultimaker doesn't provide any 32bit builds of newer Cura versions I installed Cura from your PPA cura-master.
After loading a model to print it says "Slicing..." but that's it. It does not slice, no more lines in the logfile. Logfile is attached.
[cura.log](https://github.com/thopiekar/Cura-packaging/files/1334625/cura.log)

mahowi, 2017-09-27:
FYI: Stable branch works.

thopiekar, 2017-09-28:
Has this been already reported to the Cura project?
Maybe they are able to see what is wrong and fix the issue we have in the engine.
It is a known problem and thus the software test suite is failing.

The most confusing thing for me, is, that other 32bit systems, like armhf are not failing at all. Very odd..

mahowi, 2017-09-28:
I opened [issue #2511](https://github.com/Ultimaker/Cura/issues/2511) in Cura project now.

thopiekar, 2017-09-28:
Thanks! Maybe I can manage to prepare a script, which trying to build the engine by every change (commit), which has been made on the project. Hope it will be possible to find the issue that way..

thopiekar, 2017-09-28:
(But need time to prepare this and I'm currently busy..)

smartavionics, 2017-09-28:
Hi @thopiekar , the problem here is simply the UI and engine repos were in sync when the PPA was built:

2017-09-26 21:02:34,981 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] [ERROR] Trying to retrieve unregistered setting with no value given: 'top_skin_expand_distance'

thopiekar, 2017-09-28:
That's because the test suite failed and after a while I disabled tests for i386. However, there is something that terribly broken that it won't slice on i386..

smartavionics, 2017-09-28:
Ah OK. Don't know what that could be but, in general, the master branches are working fine on my 64bit Linux system. I do see the odd synchronisation problem between repos but I think that's the price you pay for living on the bleeding edge!

thopiekar, 2017-09-28:
It exists since 2.7... it is not the price of working with the bleeding edge but the price that Ultimaker droped i386 support.

thopiekar, 2017-09-28:
I know I'm talking about my company here..

smartavionics, 2017-09-28:
The're not the only ones...

https://itsfoss.com/ubuntu-drops-32-bit-desktop/

thopiekar, 2017-09-28:
Well, I don't see why. As long as it works it is just fine and has enough RAM, I don't care. Then it is only the question about the CPU speed how much it takes.

thopiekar, 2018-02-07:
@mahowi Could you please test the latest builds in the master PPA?

Thanks!

mahowi, 2018-02-07:
@thopiekar I have installed the latest build from master PPA but it segfaults at start:
```
Fatal Python error: Segmentation fault

Thread 0x9e986b40 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x9f187b40 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0xa0289b40 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 66 in _updateThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0xa0d74b40 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0xa1575b40 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 295 in wait
  File "/usr/lib/python3.6/threading.py", line 426 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0xa1d76b40 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 295 in wait
  File "/usr/lib/python3.6/threading.py", line 426 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0xa2577b40 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 299 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1716 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1317 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0xa2d78b40 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 299 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1716 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1181 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0xa3579b40 (most recent call first):
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1102 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0xa3d7ab40 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 299 in wait
  File "/usr/lib/python3.6/threading.py", line 551 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/UM3OutputDevicePlugin.py", line 266 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0xb7c14c40 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 236 in initializeEngine
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 793 in runWithGUI
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 729 in run
  File "/usr/bin/cura", line 98 in <module>
```

 I have attached the logfile.

[cura.log](https://github.com/thopiekar/Cura-packaging/files/1703906/cura.log)

thopiekar, 2018-02-09:
I upgraded PyQt5 in the master branch. Take a look here: https://github.com/thopiekar/Cura-packaging/issues/31#issuecomment-364473874 and I hope it fixes the issue with Cura and does not break anything else.

mahowi, 2018-02-09:
I have upgraded to newest versions available on your master PPA, but now cura segfaults right after starting without writing a log or even creating its directories in user's home.

thopiekar, 2018-02-10:
Would you mind upgrading to Ubuntu bionic?

mahowi, 2018-02-10:
I will give it a try. But this could last a few days as I currently have trouble with my internet connection via DSL.

thopiekar, 2018-02-10:
No problem. Good luck getting it fixed! :crossed_fingers:

mahowi, 2018-02-18:
I'm still on 17.10 (Artful Aardvark), but the latest build seems to work now.

thopiekar, 2018-02-18:
You mean Cura 3.2.99 in the master PPA?

mahowi, 2018-02-18:
Yes, it's 3.2.99. I have updated today and it works.

thopiekar, 2018-02-18:
Can you tell me more about your hardware? Graphics card and driver? If it is really the case that the issue is now gone, then I can roll out Cura 3.2.x (stable) to the stable PPA, too.

mahowi, 2018-02-18:
It's a ThinkPad T60 with ATI graphics:

```
System:    Host: T60 Kernel: 4.13.0-32-generic i686 bits: 32
           Desktop: MATE 1.18.0  Distro: Ubuntu 17.10
Machine:   Device: laptop System: LENOVO product: 2007YML v: ThinkPad T60 serial: N/A
           Mobo: LENOVO model: 2007YML serial: N/A
           BIOS: LENOVO v: 79ETE7WW (2.27 ) date: 03/21/2011
Battery    BAT0: charge: 59.8 Wh 113.5% condition: 52.7/71.3 Wh (74%)
CPU:       Dual core Intel Core Duo T2500 (-MCP-) speed/max: 1000/2000 MHz
Graphics:  Card: Advanced Micro Devices [AMD/ATI] RV515/M54 [Mobility Radeon X1400]
           Display Server: x11 (X.Org 1.19.5 )
           drivers: ati,radeon (unloaded: modesetting,fbdev,vesa)
           Resolution: 1400x1050@60.02hz
           OpenGL: renderer: ATI RV515 version: 2.1 Mesa 17.2.8
Network:   Card-1: Intel 82573L Gigabit Ethernet Controller driver: e1000e
           Card-2: Intel PRO/Wireless 3945ABG [Golan] Network Connection
           driver: iwl3945
Drives:    HDD Total Size: 82.1GB (59.7% used)
Info:      Processes: 203 Uptime: 19 min Memory: 783.9/3025.2MB
           Client: Shell (bash) inxi: 2.3.56

```

closed 2018-02-20 by thopiekar

mahowi, 2018-02-24:
I have tried the stable branch now. On Ubuntu Artful it can't get installed because of not solvable dependencies while there are no problems with master. It searches for python-numpy-stl which is only available in Bionic as numpy-stl.
Then I have upgraded to Bionic. I can install stable without problems, but now slicing doesn't work anymore, again. The part of the window with options isn't even shown.
Master branch still works. As it's both 3.2, are there any differences?

thopiekar, 2018-02-24:
There shouldn't be. 
I'm really wondering about the numpy-stl dependency. I only set it as a recommendation. Therefore if it is not there, it should just skip it.
`The part of the window with options isn't even shown.` - You mean the sidebar?
Well, master includes some changes, which will be released in 3.3.

I will try to take a look at it and see how it comes.. :confused:

mahowi, 2018-02-24:
Yes, I mean the sidebar. I can open and resize a part, but I can't slice. Same problem as in the beginning of this thread.

Ah, yes, you're right, numpy-stl is only recommended. Unfortunately I don't remember the package which couldn't get resolved on Artful. 

Master can get installed and works on both Ubuntu versions, Artful and Bionic.