thopiekar/Cura-packaging #29: Cura crashed with Ubuntu 17.10

guyelg, 2017-10-23:
Hi, I have tried to run Cura on 2 different computer with the same error.
Cura crashed in Ubuntu 17.10 (work fine in 17.04).
Here is the relevant log:
```
2017-10-23 07:29:55,829 - DEBUG - UM.Controller.setActiveView [84]: Setting active view to SolidView
2017-10-23 07:29:58,098 - DEBUG - UM.Qt.Bindings.Theme.load [164]: Loading theme file: /usr/share/cura/resources/themes/cura/theme.json
2017-10-23 07:29:58,224 - DEBUG - UM.Qt.Bindings.Theme.load [232]: Loaded theme /usr/share/cura/resources/themes/cura
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model"
2017-10-23 07:29:58,304 - WARNING - UM.Qt.Bindings.Theme.getSize [101]: No size message_margin defined in Theme
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
2017-10-23 07:29:58,317 - WARNING - cura.Settings.ContainerManager.getContainerMetaDataEntry [184]: Could not get metadata of container  because it was not found.
QObject::connect: Parentheses expected, signal MainWindow::
Fatal Python error: Segmentation fault

Thread 0x00007f3db8a5f700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 156 in _storeStderrToLogThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f3db9260700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Backend/Backend.py", line 148 in _storeOutputToLogThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f3dba34d700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f3dbaffd700 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f3dbb7fe700 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 295 in wait
  File "/usr/lib/python3.6/threading.py", line 426 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f3dbbfff700 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 295 in wait
  File "/usr/lib/python3.6/threading.py", line 426 in acquire
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f3dc8f9b700 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 299 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1716 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1317 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f3dc979c700 (most recent call first):
  File "/usr/lib/python3.6/threading.py", line 299 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1716 in wait
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1181 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f3dc9f9d700 (most recent call first):
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1102 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f3e04660740 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 237 in initializeEngine
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 670 in run
  File "/usr/bin/cura", line 73 in <module>
Segmentation fault (core dumped)

```

thopiekar, 2017-10-23:
Yeah, I know this issue, but have no idea why it fails. To understand how it comes, I would need to setup something specific here (multiple chroots with different ubuntu releases installed). Otherwise I can't detect which ubuntu package is causing the issue here. Once known I could create an unpatched package of it. At least I guess it is caused by a patch inside a package by Canonical.

guyelg, 2017-10-26:
BTW , The AppImage working fine

Bubu, 2017-11-05:
I get the same stacktrace after upgrading to Ubuntu 17.10 and re-instralling cura.

@thopiekar Tell me If I can help debugging this problem!

thopiekar, 2017-11-07:
@Bubu Just google around "gdb python" there must be an guide how to debug Python applications on Ubuntu.
@guyelg Yes, this is known.

Bubu, 2017-11-07:
@thopiekar I ran it through gdb and I think got a bit more output:

```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007fffeb514d99 in QWindow::geometry() const () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
```

Full log including stack backtrace after the segfault here: http://code.bulix.org/rvsl5y-223667

Tell me if you want me to run any other specific commands. I guess I'd need QT debugging symbols but I'm not really sure if there is a package for that (it doesn't seem to be named libqt5gui-dbg or something similar)

thopiekar, 2017-11-07:
Yes, but all of the debugging symbols are in one big package I think.

Bubu, 2017-11-07:
Well it's more complicated than that. Ubuntu now provides automatic -dbgsym packages but you need to manually include the appropriate repos: https://wiki.ubuntu.com/Debug%20Symbol%20Packages

Anyway after hunting down a few of those dbgsym packages I get this backtrace: http://code.bulix.org/2dk4qh-223674

Excerpt:
```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007fffeb514d99 in QWindow::geometry (this=0x2702ca0) at kernel/qwindow.cpp:1633
1633	kernel/qwindow.cpp: No such file or directory.
(gdb) bt
#0  0x00007fffeb514d99 in QWindow::geometry() const (this=0x2702ca0) at kernel/qwindow.cpp:1633
#1  0x00007fffeb51a7f1 in QWindow::width() const (this=<optimized out>) at kernel/qwindow.h:226
#2  0x00007fffeb51a7f1 in QWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=<optimized out>, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>) at .moc/moc_qwindow.cpp:560
#3  0x00007fffecd0bd2b in QQmlPropertyData::readProperty(QObject*, void*) const (property=0x7fffffffcbe0, target=0x2702ca0, this=0x7fff70061bb8)
    at ../../include/QtQml/5.9.1/QtQml/private/../../../../../src/qml/qml/qqmlpropertycache_p.h:302
#8  0x00007fffc962c7f3 in  ()
```
Not sure whats now still missing for the `in ()` thing.

thopiekar, 2017-11-07:
Ok, ping me after approx 10h, so I can forward this to the Cura guys.
They have better eyes for these things or go to IRC #cura and ask there.
(They will be also available there in at least 10h :) )

Bubu, 2017-11-08:
@thopiekar Ping! :-)

I'm also in the cura irc now. (BubuIIC)

LipuFei, 2017-12-11:
Also see this: https://github.com/Ultimaker/Cura/issues/2830

lolorc, 2017-12-13:
been compiling & packaging cura myself on ubuntu 17.04 I also have this issue.
there's no libglvnd package in 17.04, so there's no -dev either.

LipuFei, 2017-12-15:
@guyelg I have the same problem when I use qt 5.8. If I switch to 5.9, the problem will go away.

thopiekar, 2017-12-15:
@LipuFei So you would say the bug is not mainly related to PyQt, but to Qt (5.8)?

LipuFei, 2017-12-16:
@thopiekar Could be, but I only have that crash if I use Qt 5.8 that I compiled. I don't have any problem with any Cura AppImage.

guyelg, 2017-12-16:
Using QT5.9. from apt-get:
> libqt5core5a is already the newest version (5.9.1+dfsg-10ubuntu1).
Cura still crash

thopiekar, 2018-01-16:
@LipuFei Upgrade to Ubuntu Bonic (18.04), which is currently under development. It uses Qt 5.9.x (https://packages.ubuntu.com/source/bionic/qtbase-opensource-src) and PyQt 5.9.x (https://packages.ubuntu.com/source/bionic/pyqt5), but still facing crashes here. The only difference is that I get a segmentation fault without a splash screen. So now something happens earlier, it seems.

LipuFei, 2018-01-17:
@thopiekar I compiled them myself, so I'm not using the debs on ubuntu ppa. Could you enable core dump and use gdb to see where it crashes?

biscuitlad, 2018-01-24:
I'm getting the same crash. I'm on Ubuntu 16.04. I get as far as adding my printer, then boom! Stack trace as above: 
```
xxx@localhost:~$ sb_release -a 
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.3 LTS
Release:        16.04
Codename:       xenial
```

thopiekar, 2018-02-07:
Could you please test the current master PPA?

On my side I'm now able to add the Ultimaker 3 as an multi-extruder machine and get no crashes.

Thanks!

ungutknut, 2018-02-07:
I guess this information here will be redundant, but maybe it's still useful in any kind (if not, sorry for spamming):

Since Xubuntu 17.10 cura-stable and master ppa refuse to start, like it's the case for many users. However the AppImage versions of 2.7.0 and 3.2.0 work just fine. So here's my terminal output when I try to start cura 3.2.0 from master-ppa:

https://pastebin.com/wxMDQZ8y

PyQt version is 5.7
Qt version is 5.9.1

If there's more information for me to share, just tell me.

Thanks btw @ thopiekar for the great work so far!

thopiekar, 2018-02-09:
An update for PyQt5 is on the way: https://code.launchpad.net/~thopiekar/+recipe/pyqt5-debian-daily
However, I don't give any guarantees when updating to it. Everything like applications and desktops, which depend on it, could break!

thopiekar, 2018-02-09:
It is there: https://github.com/thopiekar/Cura-packaging/issues/31#issuecomment-364473874

thopiekar, 2018-02-10:
Uploaded an updated package of PyQt5 to the master PPA. Upgrading from 5.7 to 5.9 didn't make a difference. I guess the issue comes from somewhere upwards (eg. Qt packages).

I'm not going to fix this. There will be Ubuntu bionic soon, which will be an LTS version and where the master packages just build and run fine.

thopiekar, 2018-02-11:
Closing this now, because I'm not going to fix this. Please upgrade to Ubuntu bionic instead.

closed 2018-02-11 by thopiekar