thopiekar/Cura-packaging #30: all multi extruder machines crash Cura

maukcc, 2017-11-27:
Every day I start my day with updating this PPA in good hopes, to see if this issue has been fixed. But sadly no. Everything else gets updated fine.
This has been going on since 2.6.99 from Juli 2017.
So for a half year now, nobody with a multi extruder machine has been able to use this PPA.
Any time frame in when this bug will be fixed?
Or is this a feature :) for some reason?

thopiekar, 2017-12-02:
Hey @maukcc ,
no it is not a "feature" 😄 The reason is that since this version I see often issues on Ubuntu.
E.g. PyQt5 is broken or something else is causing crashes.

The last days I was think about this problem here a lot and now in the end I think it is better to take a look on a port to PySide2.
The benefit I see is:
* I have already my own packages at ppa:thopiekar/pyside-git (nightly builds)
* Upgrading PyQt5 can cause issues with other applications. This way I won't touch it.

Will see how it works. Could also be that this is not easily doable as I imagine, but let's try first.

thopiekar, 2017-12-02:
Done everything to port Cura to PySide2, but still some functions are missing.

maukcc, 2017-12-04:
Great, If I can be of any help (I am not a programmer), let me know.

maukcc, 2017-12-19:
Since 3.1 we can not open Cura at all.
It will not even create the .local/share/cura folder

thopiekar, 2017-12-19:
I know, but this is nothing I can fix without an huge effort 😖

maukcc, 2018-01-16:
Update: (using the master PPA)
on Ubuntu 16.04 it installs Cura 3.0.99 and does not work
on Ubuntu 18.04 it installs Cura 3.0.3 and does work

thopiekar, 2018-01-16:
Interesting. For me it does not on Ubuntu bionic.
Could you give me more detailed info about your hardware?
Has 18.04 been installed fresh or via upgrade?
Will send you also later a command to generate a list of installed packages. I'm really curious, why running Cura still does not work for me.

maukcc, 2018-01-16:
[image]
fresh install with with update&&upgrade yesterday.
16.04 is also still installed on another partition.

thopiekar, 2018-01-16:
`dpkg-query -l > ./maukcc_all_installed_packages_on_bionic`
After that try to append the file here ( eg.  zipping might be needed ) or send the via to me by mail.

Thanks @maukcc !

maukcc, 2018-01-16:
[maukcc_all_installed_packages_on_bionic.zip](https://github.com/thopiekar/Cura-packaging/files/1635247/maukcc_all_installed_packages_on_bionic.zip)

here you go

thopiekar, 2018-01-16:
What you have installed is not my package (PPA) anymore, but one provided by Ubuntu.
Yes, Cura 3.0.3 is in Ubuntus repositories!
https://packages.ubuntu.com/de/bionic/amd64/cura

According to this page, this is done by @onitake , would be nice if he could tell us about applied patches or whatever has been done compared to the upstream sources to make it work.

onitake, 2018-01-17:
Hi @thopiekar and thanks for the heads up.
Sorry for my lack of communication. It's been a busy time for me, but I agree that more collaboration would be better for everyone.

Most of the changes that we did for Debian were related to Debian policies and a few smaller things like depending on OS packages instead of using shipped dependencies (like polyclipping).
There was one important change though, and that was removing RPATH generation. That will likely cause problems on Debian. It's also important to not intermix different package versions.

AFAIK I upstreamed all code patches that were necessary (libSavitar comes to mind), and they should be in the official repos now.

What kind of crashes do you see? I'd like to take a look, perhaps I'll notice something.

biscuitlad, 2018-01-25:
OK, I can confirm I get this crash too. If I add the K4800, it continues fine. If I add the K4800 dual extruder, boom! Dies with a seg fault. This is ridiculous - it seg faults on adding a dual nozzle printer config? What?

```
Current thread 0x000073c856ce1700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/ListModel.py", line 63 in setItems
  File "/usr/lib/python3/dist-packages/cura/Settings/ExtrudersModel.py", line 224 in __updateExtruders
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 682 in run
  File "/usr/bin/cura", line 73 in <module>
Segmentation fault (core dumped)
Running exit commands...
```
```
(xenial)xxxx@localhost:~$ lsb_release -a 
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.3 LTS
Release:        16.04
Codename:       xenial
```

onitake, 2018-01-25:
@biscuitlad I cannot reproduce this with Cura 3.1.0 and the shipped Vertex K8400 Dual profile (you meant 8400, right?). So either this was fixed in the meantime or there is compatibility issue somewhere. The segfault hints to a bug with Qt/PyQt. Python would simply generate the exception and exit.

2.7.0 is quite old, could you try a newer version?

thopiekar, 2018-01-25:
@onitake 2.7.0 is still in my stable PPA. I never had time to upgrade and then the crashes appeared. Therefore I kept it at 2.7.0. I will try to apply the RPATH patch. I don't know that much about rpath'ing, but however, as you think that this is the point where it most likely fails, then I should give it a try.
Just managed to find a way to apply your patch manually without changing the source format to 3.0 (quilt), so hope I can apply it soon.
If it is really the cause, then we should push this change hard into upstream. @onitake is there already a PR for it? And: Is there a good reason Ultimaker is using this functionality? Afaik, you made a change which removed rpath'ing completely. I guess, that Ultimaker will be only interested in a compromise, eg. a CMake option to turn it on/off.

Regards!

biscuitlad, 2018-01-25:
I can't, the latest PPA master crashes before it loads.

thopiekar, 2018-01-25:
@biscuitlad The patch @onitake mentioned is not applied now. I just made a first test.
Also the build servers at launchpad.net (Canonical) are currently very busy. It took a day to build one package, where it normally takes max half an hour for a trivial package, like Cura (so pure Cura without any additional stuff).

onitake, 2018-01-25:
RPATH allows binding external libraries to a specific path in the filesystem.
It's most useful for bundling applications into self-contained packages. That way, you can avoid linking against incompatible OS libraries at runtime.

For a platforms with proper package and dependency management (like most Linux distributions), this is counterproductive, because it may prevent loading a library managed by the OS.

I don't think this is the source of the crash described by @biscuitlad, however.

biscuitlad, 2018-01-26:
What I meant was that I couldn't test the PPA master because that crashes before even the wizard loads. 

Yes, I did mean K8400! 

I have since added my custom gcode and messed around with the settings, everything works fine. _Except_ when I try to add another printer in the form of the K8400 dual. It's bizarre - I'm assuming there's a widget or something different added to the config interface for two nozzles that causes it to crash. The config screen for K8400 for two nozzles doesn't load - it crashes on loading it.

onitake, 2018-01-27:
Have you tried deleting (or moving away) _all_ your old settings before launching Cura?

biscuitlad, 2018-01-27:
This was on a fresh install of ubuntu LTS. The only other thing that was
installed besides xfce4 was Blender 2.79. As I say, I can play around with
the single nozzle version of K8400, save settings etc. But Add Printer ->
K8400 dual -> Add = seg fault on clicking button. That's why I'm guessing
it's a widget that tries to get loaded that kills it. Mind I don't know, as
I've never seen the dual hot end config screen!

thopiekar, 2018-02-07:
@maukcc @biscuitlad Could you please test the current master PPA?

Thanks!

maukcc, 2018-02-07:
At update and upgrade we got the following packages that were kept back:
python3-arcus
python3-savitar
after sudo apt remove cura and sudo apt autoremove and re-adding the ppa and sudo apt install cura, I get the message that package cura is not available. when I do install cura*, it just installs cura without dependencies. So that does not work either.

thopiekar, 2018-02-07:
Are you sure you made a `sudo apt update`?

maukcc, 2018-02-07:
yes, that is the 1st thing I do.

thopiekar, 2018-02-07:
Good. Just asking this because I changed some package-names. How about `apt install python3-arcus`?

thopiekar, 2018-02-07:
Oh, what I haven't tested is whether installed 3rd party plugins are blocking.

maukcc, 2018-02-07:
I did a sudo apt remove cura, and now I can not re-install cura

thopiekar, 2018-02-07:
And what if you do a manual install of python3-arcus and *-savitar?

thopiekar, 2018-02-07:
Ah, damn. Looks like I removed too much when cleaning up the master PPA :confounded:

thopiekar, 2018-02-07:
[image]

maukcc, 2018-02-08:
Cura installs, and crashes when any printer is selected.
I had to delete the old .local and .config files to get it to start.
[cura.log](https://github.com/thopiekar/Cura-packaging/files/1706204/cura.log)

maukcc, 2018-02-09:
the previous remark was tested with 16.04 ubuntu.
How can I tell 18.04 not to look for onitake`s ppa but to install the thopiekar ppa?

thopiekar, 2018-02-09:
@onitake Just published his own packages on Debian and thus they are on Ubuntu, too. Just enable the master PPA and it should take the packages from there since they are newer.
Additionally, I will try to build PyQt5 packages soon, so older Ubuntu versions should work, too.

maukcc, 2018-02-09:
when just updating, we got a message that some dependancies were missing. after  a remove cura, and a fresh install it starts(also with multiextruder machines), but there is no right side, as seen in the picture.
[image]

thopiekar, 2018-02-09:
The first time I see something like this. Seems like some packages are mixed up.
What does "apt dist-upgrade" say?

maukcc, 2018-02-09:
it says: nothing to upgrade :)

thopiekar, 2018-02-09:
Strange 😕 
Could you do this again: https://github.com/thopiekar/Cura-packaging/issues/30#issuecomment-357957223

maukcc, 2018-02-09:
here you go:
[maukcc_all_installed_packages_on_bionic.txt](https://github.com/thopiekar/Cura-packaging/files/1710771/maukcc_all_installed_packages_on_bionic.txt)

maukcc, 2018-02-09:
these are the last lines when I install onitake and then upgrade to topiekar:

> Selecting previously unselected package cura-resources-firmware.
> Preparing to unpack .../16-cura-resources-firmware_3.2.0~201802071002~rev86~pkg196~ubuntu18.04.1_all.deb ...
> Unpacking cura-resources-firmware (3.2.0~201802071002~rev86~pkg196~ubuntu18.04.1) ...
> Selecting previously unselected package python3-colorlog.
> Preparing to unpack .../17-python3-colorlog_3.1.2-1_all.deb ...
> Unpacking python3-colorlog (3.1.2-1) ...
> Errors were encountered while processing:
>  /tmp/apt-dpkg-install-Qsb1lC/03-cura-resources-materials_3.2.0~201802071001~rev210~pkg196~ubuntu18.04.1_all.deb
> E: Sub-process /usr/bin/dpkg returned an error code (1)

maukcc, 2018-02-09:
and this when I do : sudo apt --fix-broken install
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> Correcting dependencies... Done
> The following packages were automatically installed and are no longer required:
>   fdm-materials libpolyclipping22 libpugixml1v5
> Use 'sudo apt autoremove' to remove them.
> The following additional packages will be installed:
>   cura-resources-materials
> The following NEW packages will be installed:
>   cura-resources-materials
> 0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
> 17 not fully installed or removed.
> Need to get 0 B/13,1 kB of archives.
> After this operation, 260 kB of additional disk space will be used.
> Do you want to continue? [Y/n]

thopiekar, 2018-02-09:
And what if you do this? Agreeing with [y]?

maukcc, 2018-02-09:
> Do you want to continue? [Y/n] y
> (Reading database ... 177464 files and directories currently installed.)
> Preparing to unpack .../cura-resources-materials_3.2.0~201802071001~rev210~pkg196~ubuntu18.04.1_all.deb ...
> Unpacking cura-resources-materials (3.2.0~201802071001~rev210~pkg196~ubuntu18.04.1) ...
> dpkg: error processing archive /var/cache/apt/archives/cura-resources-materials_3.2.0~201802071001~rev210~pkg196~ubuntu18.04.1_all.deb (--unpack):
>  trying to overwrite '/usr/share/cura/resources/materials/chromatik_pla.xml.fdm_material', which is also in package fdm-materials 3.1.0-1
> Errors were encountered while processing:
>  /var/cache/apt/archives/cura-resources-materials_3.2.0~201802071001~rev210~pkg196~ubuntu18.04.1_all.deb
> E: Sub-process /usr/bin/dpkg returned an error code (1)
>

thopiekar, 2018-02-09:
Ah! Ok, @onitake is using different package name for the material definitions. Will try to resolve this package conflict.

onitake, 2018-02-09:
I simply used the same package name as the upstream project (fdm-materials).

thopiekar, 2018-02-09:
@onitake: But it installs files into /usr/share/cura/resources/materials. That's why in my imagination it should be called something like cura-resources-materials.
If it would install all the files in a common place like /usr/share/fdm-materials, I would agree.

onitake, 2018-02-09:
Ok, that is true. But doing that would have required changing the default path in Cura as well.
I don't think it's that big a deal, but the conflict is unfortunate.

We can resolve this either by agreeing on a common name or adding mutual Conflicts: lines to debian/control. I'm a bit reluctant to change the package name now, though...

thopiekar, 2018-02-09:
Yeah, no problem. That's why I still say inside the package that it provides fdm-materials. I can also imagine that other projects in the future can reuse the info inside these XML files.
For now I will keep my convention 😉

onitake, 2018-02-09:
Here's some more info on the possibilities: https://wiki.debian.org/PackageTransition

thopiekar, 2018-02-10:
@maukcc The package conflict should be resolved now. Could you please check whether Cura runs fine now? (With sidebar?)

maukcc, 2018-02-10:
update && upgrade works fine now. but the same result: no right side sidebar.
I also did a fresh install with the same results. here the packages before and after install:
[maukcc_all_installed_packages_on_bionic_after_cura_install.txt](https://github.com/thopiekar/Cura-packaging/files/1713351/maukcc_all_installed_packages_on_bionic_after_cura_install.txt)
[maukcc_all_installed_packages_on_bionic_before_cura_install.txt](https://github.com/thopiekar/Cura-packaging/files/1713352/maukcc_all_installed_packages_on_bionic_before_cura_install.txt)

It also will not let me load in .stl files

thopiekar, 2018-02-10:
Fresh install in meaning of reinstalling Ubuntu*?

maukcc, 2018-02-10:
No, nothing as dramatic as that :)
apt remove cura
apt autoremove

thopiekar, 2018-02-10:
Ok, did you try to remove all settings? (To remember: ~/.local/share/cura/3.2/*)
I'm really curious how it comes that the sidebar is missing. Currently Windows 10 is running here, because of some development work. Will reboot into Kubuntu asap and check myself.

maukcc, 2018-02-10:
yes, .config and .local files were removed

thopiekar, 2018-02-10:
Strange 😕 
Could you send the log of Cura then?

maukcc, 2018-02-10:
[cura.log](https://github.com/thopiekar/Cura-packaging/files/1713377/cura.log)

thopiekar, 2018-02-10:
The reason for your failed load:
```
2018-02-10 13:12:59,086 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:     self.load_file(file_name, mesh_builder, _use_numpystl = use_numpystl)
2018-02-10 13:12:59,086 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:   File "/usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py", line 35, in load_file
2018-02-10 13:12:59,087 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:     self._loadWithNumpySTL(file_name, mesh_builder)
2018-02-10 13:12:59,087 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:   File "/usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py", line 87, in _loadWithNumpySTL
2018-02-10 13:12:59,087 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:     for loaded_data in stl.mesh.Mesh.from_multi_file(file_name):
2018-02-10 13:12:59,087 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:   File "/usr/lib/python3/dist-packages/stl/stl.py", line 355, in from_multi_file
2018-02-10 13:12:59,088 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:     raw_data = cls.load(fh, mode=mode, speedups=speedups)
2018-02-10 13:12:59,088 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:   File "/usr/lib/python3/dist-packages/stl/stl.py", line 81, in load
2018-02-10 13:12:59,088 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:     name, data = cls._load_binary(fh, header, check_size=False)
2018-02-10 13:12:59,088 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:   File "/usr/lib/python3/dist-packages/stl/stl.py", line 108, in _load_binary
2018-02-10 13:12:59,089 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]:     count, MAX_COUNT)
2018-02-10 13:12:59,089 - ERROR - [(140205412914944)-Thread-3] UM.Logger.logException [86]: AssertionError: File too large, got 1869573152 triangles which exceeds the maximum of 10000000
```
Looks to me like NumPy-STL can't handle your file. But I don't see any errors about the sidebar. How does Cura behave when using a different printer? Eg. Ultimaker 3?

maukcc, 2018-02-10:
With UM3 (or any other) it reacts the same.

thopiekar, 2018-02-11:
@maukcc Could you please use the stable PPA now?
The packages built on top of 3.2 branches are online and I recommend the tool "ppa-purge" to remove the master PPA.

Regards!

[18 more replies]

closed 2018-02-20 by thopiekar