thopiekar/Cura-packaging #35: Cannot install in Ubuntu Xenial libprotobuf version error

martinberlin, 2018-02-11:
Hello anyone can give me some hint on how to upgrade this lib ?

I did the following:
Uninstall old version of libprotobuf11 

sudo add-apt-repository ppa:thopiekar/cura
And then after installing I get after launching cura:
[libprotobuf FATAL google/protobuf/stubs/common.cc:67] This program requires version 3.5.0 of the Protocol Buffer runtime library, but the installed version is 3.1.0.  Please update your library.

But it seems that even uninstalling old libprotobuf version, when installing cura, still grabs an old version of this lib. Thanks for any guidance on how to solve this!

thopiekar, 2018-02-11:
Moin @martinberlin,

could you please tell me why you want to uninstall this package?
It is needed for arcus->cura. Does it conflict with your projects?

martinberlin, 2018-02-12:
The thing is in another Ubuntu machine I installed your Cura packages and it worked out perfect. No problems at all. 
And I wanted to repeat this on my laptop but I keep on getting this error message. I really don't know how to clean it. Must be there is some reference to this libprotobuf library somewhere and it's driving me mad.

I wanted to uninstall it only to reinstall the right versions and make it work. But I need to understand what's the problem first. Your package works with 3.1.0 of libprotobuf so I don't really understand why in this machine is looking for a higher version. 
Thanks a lot for packaging this. I compiled other things before but fighting ours to get this running is not my thing ! Your builds are great 👍

martinberlin, 2018-02-12:
OK I know what can be happening! Months before I tried to compile Cura on this machine without success. So I clean up all dependencies and make a fresh install with your package.

thopiekar, 2018-02-12:
Haha, no problem. Keep up your experiments, whatever they are 😄 
Cura does not have high requirements on protobuf. Therefore I kept this version and currently don't have the time to upgrade it to something more recent. I just hope Ubuntu comes (or already is) with newer protobuf versions in their software repositories.

closed 2018-02-12 by thopiekar

martinberlin, 2018-02-13:
Last little help please! Cleaned up all references to libarcus/libprotobuf and got CuraEngine running. 
Now the error that comes is:

$ xtrace cura
No display name to create specified, trying : 9

And dies there! GRR!! @thopiekar do you think I can try something else before installing all Ubuntu again to make this work ?
Thank you enormoulsy  :)  And greetings from Berlin

thopiekar, 2018-02-13:
I don't know. I would say that it comes from xtrace. Never used that tool before, but you can in most cases just start Cura in the shell via "cura".