thopiekar/Cura-packaging #36: Found an error installing this on fresh Ubuntu 16 install

martinberlin, 2018-02-13:
Hallo, 
this time I'm trying this on a very fresh Ubuntu install, nothing has been touched. No

[non-master]
$ cura
cura.CrashHandler.show [39]: An uncaught exception has occurred!
cura.CrashHandler.show [42]: Traceback (most recent call last):
cura.CrashHandler.show [42]:   File "/usr/bin/cura", line 54, in <module>
cura.CrashHandler.show [42]:     import cura.CuraApplication
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 7, in <module>
cura.CrashHandler.show [42]:     from UM.Qt.QtApplication import QtApplication
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 15, in <module>
cura.CrashHandler.show [42]:     from UM.FileHandler.ReadFileJob import ReadFileJob
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/FileHandler/ReadFileJob.py", line 4, in <module>
cura.CrashHandler.show [42]:     from UM.Job import Job
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/Job.py", line 8, in <module>
cura.CrashHandler.show [42]:     from UM.JobQueue import JobQueue
cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 10, in <module>
cura.CrashHandler.show [42]:     from typing import TYPE_CHECKING, List, Callable, Any
cura.CrashHandler.show [42]: ImportError: cannot import name 'TYPE_CHECKING'

[master]
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 662, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 21, in <module>
    from UM.Application import Application
  File "/usr/lib/python3/dist-packages/UM/Application.py", line 27, in <module>
    from typing import TYPE_CHECKING, List, Callable, Any, Optional
ImportError: cannot import name 'TYPE_CHECKING'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/cura", line 128, in <module>
    import cura.CuraApplication
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 8, in <module>
    from UM.Qt.QtApplication import QtApplication
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 15, in <module>
    from UM.FileHandler.ReadFileJob import ReadFileJob
  File "/usr/lib/python3/dist-packages/UM/FileHandler/ReadFileJob.py", line 4, in <module>
    from UM.Job import Job
  File "/usr/lib/python3/dist-packages/UM/Job.py", line 8, in <module>
    from UM.JobQueue import JobQueue
  File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 10, in <module>
    from typing import TYPE_CHECKING, List, Callable, Any
ImportError: cannot import name 'TYPE_CHECKING'

After doing the whole app dist-upgrade thing comes another different error:
Segfault happened at: 
source (%rax) not located in a known VMA region
destination "%xmm4" ok

martinberlin, 2018-02-13:
My question is: 
On what fresh install I can have this running ? What distribution ?
Or maybe Debian ?  Before I had this running, maybe it was because I had an older ubuntu version ?

The problem is that I cannot see exactly what is going on. This guys of ultimaker made something really ugly to work in Linux. Otherwise it cannot be so hard to build, I mean another programs like Slic3r work without any issues ... What does cura to have this complexity on being installed ?

thopiekar, 2018-02-13:
I have no idea, what's wrong with Cura together with Ubuntu. The problem occurs somewhere in the depth. I can only guess it is one of the dependencies Cura has.
What I can only suggest, is installing Ubuntu Bionic 18.04, which is going to be released in 2 months. This one is working here :+1: 

The segfaults just occurred during my internship at Ultimaker, where I was very busy and had no time to debug it the first time I saw it.

martinberlin, 2018-02-13:
Ahh great, but then I can already try this right ?
http://cdimage.ubuntu.com/daily-live/current/HEADER.html

This is my 3D printer laptop so I don't really mind to start over again. Thanks for the hint Thomas!

thopiekar, 2018-02-14:
Yes, this is the right location.
Alright, and since Bionic is going to be an LTS (long time support) release, you won't need to upgrade too often. 😉

closed 2018-02-20 by thopiekar