thopiekar/Cura-packaging #38: Cura will only load ascii .stl files

maukcc, 2018-02-20:
Cura will only load ascii .stl files, binary .stl files generate an error.

[dualXYtest1_ascii.stl.txt](https://github.com/thopiekar/Cura-packaging/files/1740196/dualXYtest1_ascii.stl.txt)
[dualXYtest1_binary.stl.txt](https://github.com/thopiekar/Cura-packaging/files/1740198/dualXYtest1_binary.stl.txt)

thopiekar, 2018-02-20:
I guess this is due to missing Numpy-STL package.
But it is available in upstream as: python3-stl
https://github.com/thopiekar/Cura-packaging/commit/45ce9ce0567acfd82ffbe78c35874545ecca3eb2

Ghostkeeper, 2018-03-12:
Unable to reproduce. I've disabled the numpy-stl package running from source, but the back-up still loaded binary STL files fine.

maukcc, 2018-03-13:
Did I make a mistake here? because I too can open the binary file, but not the ascii files.
@Ghostkeeper : can you try the ascii file as well?

Ghostkeeper, 2018-04-11:
The ASCII file also works for me, at least in the current state of Master.

[image]

maukcc, 2018-08-22:
Yes, binary files can be loaded, ascii files can not.

thopiekar, 2018-08-22:
Maybe related :question: 
https://github.com/Ultimaker/Uranium/pull/393

maukcc, 2018-10-08:
This is still present.
I removed Cura and  installed from the debian repositories (3.1) and that worked.
Then updated to the stable version of this ppa and lost the ability to load the .stl files
The only change I noticed apart from adding a bunch of new files was the deletion of libpolyclipping22.
Maybe that has something to do with it ?

thopiekar, 2018-10-13:
Well, I think it is not slicing because of #50 
Has been a long time ago since I made my last slice on my Linux installation.

thopiekar, 2018-11-01:
Found also issues with the STLReader. For some reason the NumPy-STL module fails to read files.
Maybe the NumPy-STL package in Ubuntu is outdated.

However, added a workaround to STLReader to be able to fallback.

PR will follow soon.

closed 2018-11-04 by thopiekar