thopiekar/Cura-packaging #39: Octoprint plugin crashes Cura

maukcc, 2018-02-21:
The plugin install correct but, when trying to connect, Cura crashes
Here is the log
[cura.log](https://github.com/thopiekar/Cura-packaging/files/1743410/cura.log)

thopiekar, 2018-02-21:
@fieldOfView FYI
@maukcc I don't see any reference to an old plugin package, but did you make sure all 3rd-party plugins are removed from apt/dpkg?

fieldOfView, 2018-02-21:
If you use the master branch of Cura, you must also use the master banch of the OctoPrintPlugin

thopiekar, 2018-02-21:
@maukcc: Based on @fieldOfView 's info, please remove cura and all dependencies from the master PPA and install Cura from the stable PPA again.

maukcc, 2018-02-21:
Using the master branch of OctoPrintPlugin worked.
I am just not seeing the video feed

closed 2018-02-21 by maukcc