thopiekar/Cura-packaging #40: Missing dependency

kvdveer, 2018-02-28:
On an (almost) clean install of Ubuntu 18.04 (currently unreleased), Cura from the (non-master) PPA will not show its sidebar. The log shows:

```
2018-02-28 20:35:37,695 - WARNING - [(140036139181888)-MainThread] UM.Qt.QtApplication.__onQmlWarning [267]: file:///usr/share/cura/resources/qml/Sidebar.qml:5:1: module "QtQuick.Controls" version 2.1 is not installed
```

installing `qml-module-qtquick-controls2` seems to resolve the issue.

thopiekar, 2018-02-28:
Hmm it should be there:
https://github.com/thopiekar/Cura-packaging/blob/3.2/Cura/control#L30

kvdveer, 2018-03-01:
That's weird. This is what my machine reports:

```
kvdveer@kvdveer-laptop:~$ apt-cache show cura
Package: cura
Priority: optional
Section: graphics
Installed-Size: 45142
Maintainer: Thomas Karl Pietrowski <thopiekar@googlemail.com>
Architecture: all
Version: 1:3.2.0~201802201608~rev2754~pkg200~ubuntu18.04.1
Recommends: fonts-open-sans, cura-resources-firmware, cura-plugins-all, python3-numpy-stl
Provides: cura-plugin-api-3
Depends: python3:any (>= 3.4~), cura-engine, cura-resources-materials, python3-uranium, python3-savitar, qml-module-qtqml-models2, qml-module-qtquick-controls, qml-module-qtquick-dialogs, qml-module-qt-labs-settings, qml-module-qt-labs-folderlistmodel
Filename: pool/main/c/cura/cura_3.2.0~201802201608~rev2754~pkg200~ubuntu18.04.1_all.deb
Size: 7481640
MD5sum: 548733c596b46feff2c367a10748db0a
SHA1: 5cf259149ba151951350985b4c120aed6673c3e9
SHA256: 54b6f8b41b61daa8e2a4683e37b89b64f17e1ce59252b87eb7aae40246ddd131
Description-en: 3D printer / slicing GUI built on top of the Uranium framework
 This is the new, shiny frontend for Cura. daid/Cura is the old legacy Cura that everyone knows and loves/hates.
 .
 We re-worked the whole GUI code at Ultimaker, because the old code started to become a unmaintainable.
Description-md5: cc29ecba270034004b1ef461747e881d
Python3-Version: ,
```  

That dependency list is different from the control file. I'm seeing the same discrepancy [here](http://ppa.launchpad.net/thopiekar/cura/ubuntu/dists/bionic/main/binary-amd64/). This would suggest that the build process is not properly picking up the dependency.

thopiekar, 2018-03-01:
Good catch! The mode of the build recipe was still on "Build on request". Therefore the changes I made didn't trigger a new build.

https://code.launchpad.net/~thopiekar/+recipe/cura-stable-daily

Thanks!

closed 2018-03-01 by thopiekar