thopiekar/Cura-packaging #45: 3.3.1 speichert Dateien nicht mehr

herrdeh, 2018-07-09:
Siehe hier: [https://github.com/Ultimaker/Cura/issues/4042](url)
.
Seit dem letzten Systemupdate kann ich weder gcode noch curaproject speichern. Über die 3.4 würde ich mich seeeehr freuen !  (-:

Ganz lieb bitte....
Wolf

thopiekar, 2018-07-09:
Hallo Wolf,

gerne schau ich mir das an. Allerdings habe ich nicht viel Zeit, daher verspäten sich diese Dinge immer etwas. Andererseits ist dies auch nicht schlecht, denn dann kann ich oft auf die nächste Unterversion zur Fehlerbehebung ansetzen. Denke das ist auch was Leute von der (stable) PPA erwarten.

I tried to reproduce your issue..
[image]
and couldn't reproduce it. Once I can see what the error messages are within your logs at https://github.com/Ultimaker/Cura/issues/4042 and see that this is not related to your computer (messed up file/directory permissions), then I will care about upgrading to 3.4.x earlier.

herrdeh, 2018-07-10:
Hm. After a first launch of 3.4 now 3.3.1 can save again... Strange. 
Anyway - my problem is solved - thanks for attention and help!

closed 2018-07-10 by herrdeh