thopiekar/Cura-packaging #46: Cura version on Ubuntu Software out of date, does not auto update

mrjohnc, 2018-08-06:
Hi Thomas

Thanks very much for adding Cura to Ubuntu Software, I noticed the version available is out of date and also does not auto update, I asked the developers about this is and discovered you added it. Please could you readd the auto update feature back in? 

Discussion with developers here
https://github.com/Ultimaker/Cura/issues/4197#issuecomment-410679438

Thanks very much

thopiekar, 2018-08-06:
Auto update is there, if a version tag gets updated.
But whenever I make an update I need to change the version number in the packing files manually. This is nothing that can be automated by launchpad.net. At least as far as I know 😅

I'll put it onto my list 😉👍

Thank you for reminding.
- Thomas

mrjohnc, 2018-08-06:
Thanks very much for the quick response, sorry this is a manual task :( I don't really understand about the version tags, hopefully it can be fixed so you don't have to do it manually forever :) 

Best

John

thopiekar, 2018-08-21:
3.4 is up. I'll bump the version from 3.4.0 to 3.4.1 now.
However, 3.4.1 should be used all the time. Only the splash screen will show different :wink: