thopiekar/Cura-packaging #48: Adding python3-shapely

thopiekar, 2018-08-19:
- [ ]  Adding initial package

As explained here: https://github.com/Ultimaker/Uranium/pull/375

norro, 2020-01-03:
Current cura stable not installable for me on ubuntu 19.10 due to erroneous dependency to package `python3-trimesh`, which does not seem to be existing for ubuntu 19.10.

`The following packages have unmet dependencies:`
` cura : Depends: python3-uranium but it is not going to be installed`
`        Recommends: python3-trimesh but it is not installable`

thopiekar, 2020-01-06:
.. and it is not getting better. Cura is getting to be a huge dependency monster.
I already packaged it and tested it in the master PPA.
Since I never had problems, I'll drop it into stable now.

@norro Thank you for the reminder :+1:

thopiekar, 2020-01-06:
Wait.. Trimesh is optional. The problem mentioned in your message is because of uranium. Something is missing there.
Can you run `apt install python3-uranium`

norro, 2020-01-08:
`The following packages have unmet dependencies:`
` python3-uranium : Depends: python3-arcus but it is not going to be installed`

norro, 2020-01-08:
python3-arcus seems to be the problem:
`The following packages have unmet dependencies:`
` python3-arcus : Depends: libarcus3 (= 1:4.4~202001021825~rev172~pkg327~ubuntu19.10.1) but 1:4.5~202001022332~rev174~pkg330~ubuntu19.10.1 is to be installed`

thopiekar, 2020-01-11:
Looks like there is libarcus left from an installation based on master.
Try to clean up all the packages installed from the master and stable PPA and try again.

norro, 2020-01-25:
Tpo me it seems that it is simply missing the `python3-trimesh` package:

```
$ sudo apt install python3-trimesh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package python3-trimesh is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python3-trimesh' has no installation candidate
```

I have only your master ppa active on my PC:
`deb http://ppa.launchpad.net/thopiekar/cura/ubuntu eoan main`