thopiekar/Cura-packaging #50: Incompatible libprotobuf

shernshiou, 2018-10-01:
The ppa is using libprotobuf that comes from Ubuntu 18.04 which is version 3.0.0. Since the cura-engine needs version 3.5, is it possible that  this ppa bundle version 3.5 as well?

```
2018-10-01 17:08:30,595 - DEBUG - [Thread-16] UM.Backend.Backend._backendLog [94]: [Backend] [libprotobuf FATAL google/protobuf/stubs/common.cc:67] This program requires version 3.5.0 of the Protocol Buffer runtime library, but the installed version is 3.0.0.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "google/protobuf/any.pb.cc".)
2018-10-01 17:08:30,596 - DEBUG - [Thread-16] UM.Backend.Backend._backendLog [94]: [Backend] terminate called after throwing an instance of 'google::protobuf::FatalException'
```

thopiekar, 2018-10-01:
Thank you for reporting this. I'm recently slicing that often really.
I already prepared an updated package in the past for previous Ubuntu versions. Can do the same here, but can't guarantee that I will be able to do this soon. I'm really busy and this kind of stuff needs time and nerves. Otherwise I mess up the PPA and that again takes more time.

shernshiou, 2018-10-02:
Thanks @thopiekar

thopiekar, 2018-11-01:
Works here again. Therefore closing the ticket.
If you still have problems, just let me know and reopen it :wink:

closed 2018-11-01 by thopiekar