thopiekar/Cura-packaging #54: Cura fails to run on 18.04

battlezone, 2018-11-24:
I just installed Cura via your PPA, but when I start it, nothing shows up.
When running it from terminal I get the following:
`Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 21, in <module>
    from UM.Application import Application
  File "/usr/lib/python3/dist-packages/UM/Application.py", line 15, in <module>
    from UM.Settings.ContainerRegistry import ContainerRegistry
  File "/usr/lib/python3/dist-packages/UM/Settings/ContainerRegistry.py", line 36, in <module>
    import UM.Qt.QtApplication
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 15, in <module>
    from UM.FileHandler.ReadFileJob import ReadFileJob
  File "/usr/lib/python3/dist-packages/UM/FileHandler/ReadFileJob.py", line 5, in <module>
    from UM.Application import Application
ImportError: cannot import name 'Application'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/cura", line 135, in <module>
    from cura.CuraApplication import CuraApplication
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 16, in <module>
    from UM.Application import Application
  File "/usr/lib/python3/dist-packages/UM/Application.py", line 15, in <module>
    from UM.Settings.ContainerRegistry import ContainerRegistry
  File "/usr/lib/python3/dist-packages/UM/Settings/ContainerRegistry.py", line 36, in <module>
    import UM.Qt.QtApplication
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 15, in <module>
    from UM.FileHandler.ReadFileJob import ReadFileJob
  File "/usr/lib/python3/dist-packages/UM/FileHandler/ReadFileJob.py", line 5, in <module>
    from UM.Application import Application
ImportError: cannot import name 'Application'
`

This is both the case for cura and cura-master PPA

thopiekar, 2018-11-26:
Think I found the problem.
Let's see..

thopiekar, 2018-11-26:
Can you try again? Thanks!

closed 2018-11-27 by thopiekar

battlezone, 2018-11-27:
Still not working. Same error.

battlezone, 2018-12-03:
It seems there was a delay with updating the ppa. I noticed there was a cura update on saturday/sunday and it's indeed working fine now. Thank you!

thopiekar, 2018-12-03:
Haha, was only experimenting this day. Actually thought that I ended up
with the same problem. But great to hear that it works.
I won't touch the stable PPA anymore then 😂