thopiekar/Cura-packaging #55: Stable version 3.6.0 fails to slice

neildarlow, 2018-11-27:
It appears that the latest Cura stable packaging at version 3.6.0 contains a cura-engine package at version 3.4.1.

When you attempt to slice a STL file there is no activity reported on the slicing progress bar and the UI remains at Slicing...

The AppImage of the same version works correctly.

Tested on Linux Mint 19 Cinnamon Edition.

neildarlow, 2018-11-27:
List of available packages shown by Synaptic. Note the highlighted cura-engine version.

[image]

thopiekar, 2018-11-27:
Good catch. I missed to update the protobuf package.
Triggered to build package for the stable PPA. Should be fixed soon.

thopiekar, 2018-11-28:
Currently blocked because of this:
https://github.com/Ultimaker/Cura/issues/4900

Nihlus, 2018-11-28:
Is there a way to roll back to the previous version? I'm not seeing any previous versions in the PPA.

thopiekar, 2018-11-28:
No there is no way to do that. I hope I get a reply on this that it is safe
to use the code from the master branch, which leads to no build errors.

neildarlow, 2018-11-28:
@Nihlus Until this is resolved you can use the AppImage download from Ultimaker. This is what I am currently doing.

@thopiekar Does a comparison of the stable and master build recipies provide any clues? Could be another added dependency?

thopiekar, 2018-11-28:
@neildarlow Well, I tried to build the engine in the master PPA (so all dependencies should be the same), but ended out with the same error. Thus, there must be something with the code released as 3.6.
I mean since the stable PPA is broken anyways now, I can just build the engine from master and see what happens. If it runs, then we have something working at least.
So then all code will be 3.6, but the engine will be newer with the needed fix.

What do you think?

neildarlow, 2018-11-28:
@thopiekar That sounds reasonable provided we can drop back to stable once things settle. Newer dependency versions in Master could impact that. It is worth trying.

thopiekar, 2018-11-28:
The engine is there. Can you update and see whether it works?

neildarlow, 2018-11-28:
Slicing starts and gets to about 50% then stops. Not successful unfortunately.

Nihlus, 2018-11-29:
Yeah, same here. Is it possible for you to keep one or two older versions in the PPA in the future, so we can avoid lockups like this?

thopiekar, 2018-11-29:
Well, I never had these unexpected problems before.
Making this sure would mean that I would have to create another stable PPA for backup. As a consequence this would mean more maintenance and more work.
However, I don't have the time for this and actually need to spend more time on the plugins right now.

thopiekar, 2018-11-29:
Sure, but I never removed anything, but launchpad.net does a clean up on
its own. Otherwise they won't be able to handle all the data created by
daily builds..

tobie-de-beer, 2018-12-20:
I managed to get all the 3.4 files I needed from the apt cache to get 3.4 up and running again.....

thopiekar, 2018-12-20:
Great, I will probably update it soon to the 4.0 beta due to the issues here.
Has anyone tried the ETA, which is currently in the master PPA?

Thanks!

thopiekar, 2018-12-26:
Updated to 4.0 - better having a beta, then non-working software.

Merry Christmas!

closed 2018-12-26 by thopiekar