thopiekar/Cura-packaging #58: UI does not appear

ianrumford, 2018-12-30:
Hi Thomas,

Thanks for providing the ppa. But cura doesn't work for me.

The install completes successfully.

When I start cura, I see the splash screen and the "I understand and agree" pane but no UI appears. (btw the text box of the "agree" pane is empty)

I've tried on two instances of 18.04, one a completely new 18.04 instance running under VirtualBox.

It looks like cura has started and it talking to the engine ok: ps shows

> ian       4451  4398 28 14:44 pts/1    00:00:02 /usr/bin/python3 /usr/bin/cura
> ian       4477  4451  0 14:44 pts/1    00:00:00 /usr/bin/CuraEngine connect 127.0.0.1:49674

Starting cura form a bash shell, I see a number of errors, notably about loading the cura-light theme (from /usr/share/cura/resources/themes/cura-light)

>  ERROR - [MainThread] UM.Qt.Bindings.Theme.load [242]: file:///usr/share/cura/resources/themes/cura-light/styles.qml:4:1: module "QtQuick" version 2.10 is not installed

According to [this page](http://doc.qt.io/qt-5/qtquickcontrols-index.html) QT 5.9 uses QtQuick 2.9 and QtQuick.Controls 2.2. I can find nothing about the Styles version though.

I have edited styles.qml to try these and other values but the UI never appears.

Any light you can throw on this would be most helpful.

Thanks in advance.

btw I have successfully run cura 3.1 from the Canonical repos.

bmn-m, 2018-12-30:
Same here. Version is 4.0~201812272158~rev3793~pkg293~ubuntu18.04.1

bmn-m, 2018-12-30:
Thank you very much for all your efforts. Great Job!!! The cura.appimage is a huge mess and i really like your PPA Version.

ianrumford, 2018-12-31:
Hi Thomas, 

Thanks for the response.

If I can test anything, please do ask: very easy for me to spin up my VirtualBox test machine and try something out.

thopiekar, 2018-12-31:
Which desktop environment are you using? KDE?

bmn-m, 2018-12-31:
yes Ubuntu 18.04.1

ianrumford, 2018-12-31:
Yep Bionic 18.04.1 with all updates applied

thopiekar, 2018-12-31:
Ok, I found PPAs with Qt 5.11, but don't know whether it is risky to upgrade to it or not :(

thopiekar, 2018-12-31:
Will try it out in my VM. Give me some time..
Sadly my laptop is busy with building some stuff.

But yeah, Ultimaker is using Qt 5.10 in their downloads. Missed this move they made in cura-build-environment :(

ianrumford, 2018-12-31:
Thanks! Let us know if we can share the load.

thopiekar, 2018-12-31:
Yes, you can! Make a snap in your VBox and try these PPAs:
* https://launchpad.net/~packetlost/+archive/ubuntu/qt-opensource
* https://launchpad.net/~panfaust/+archive/ubuntu/qt2

Try both separately and see whether Cura and (most importantly) other software works still, too.
Don't want to suggest people to add one of these PPAs, which might break their system.
I guess the worst case would be running KDE after such an upgrade since it depends on many Qt modules and libraries..

marcusj, 2018-12-31:
Same problem here on 18.04.  Silly question, can I just roll back to your penultimate release so I can get slicing again? (If so, some hint, please.)

thopiekar, 2018-12-31:
Nope, rolling back is not possible... at least not easily.
What you can do for sure is uninstalling the PPA and installing Ubuntu's older version.
But I guess that using an AppImage temporarily is the fastest solution.

thopiekar, 2018-12-31:
@ianrumford Are you investigating on it?

ianrumford, 2018-12-31:
Jest back. Will have a go in next hour.

thopiekar, 2018-12-31:
No hurries.. If you don't manage it is no problem.
I will then continue looking for a solution then. :wink:

ianrumford, 2018-12-31:
Thomas,

The first of those ppas (qt-opensource) only has a package for trusty.

Trying the second ...

ianrumford, 2018-12-31:
The 2nd ppa causes Ubuntu to says not all updates can be installed and offers a partial upgrade.

25 no longer needed packages. 1 to be removed (cura!), 28 to be upgraded.
partial upgrade completed ok.

trying to reinstall cura and got:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 cura : Depends: python3-uranium but it is not going to be installed
        Depends: qml-module-qtquick-controls but it is not going to be installed
        Depends: qml-module-qtquick-controls2 but it is not going to be installed
        Depends: qml-module-qtquick-dialogs but it is not going to be installed
        Recommends: python3-pyqt5 but it is not going to be installed
        Recommends: python3-pyqt5.qtwebsockets but it is not going to be installed
        Recommends: python3-pyqt5.qtx11extras but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

pass on this one for moment. will try the third one.

thopiekar, 2018-12-31:
Hmm. Looks like it breaks something in the PyQt packages then.
Well, as long as it don't break other things I could take this Qt version and rebuild PyQt on top of it. Shouldn't take toooo much time.. :thinking:

ianrumford, 2018-12-31:
Third one (qt3) again causes a partial upgrade.

16 no longer need, 1 to be removed (cura), 35 packages to be upgraded.

partial upgrade completes but get same error trying to reinstall cura:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 cura : Depends: python3-uranium but it is not going to be installed
        Recommends: python3-pyqt5 but it is not going to be installed
        Recommends: python3-pyqt5.qtwebsockets but it is not going to be installed
        Recommends: python3-pyqt5.qtx11extras but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

not good.

ianrumford, 2019-01-03:
Hi Thomas,

Hope you are well.  

An update to cura from the ppa came though for me today.  The "QtQuck 2.10" error no longer appears in the log but the UI still does not appear.

Still seeing lines like these in the log before the load of the cura-light theme.

> QQmlApplicationEngine failed to load component
> file:///usr/share/cura/resources/qml/Cura.qml:139 Type MainWindowHeader unavailable
> file:///usr/share/cura/resources/qml/MainWindow/MainWindowHeader.qml:119 Type AccountWidget unavailable
> file:///usr/share/cura/resources/qml/Account/AccountWidget.qml:57 Type AccountDetails unavailable
> file:///usr/share/cura/resources/qml/Account/AccountDetails.qml:61 Type UserOperations unavailable
> file:///usr/share/cura/resources/qml/Account/UserOperations.qml:14 Type Cura.SecondaryButton unavailable
> file:///usr/share/cura/resources/qml/SecondaryButton.qml:10 Type Cura.ActionButton unavailable
f> ile:///usr/share/cura/resources/qml/ActionButton.qml:92 Type Cura.RoundedRectangle unavailable
> file:///usr/share/cura/resources/qml/RoundedRectangle.qml:22 Unexpected token `reserved word'
> file:///usr/share/cura/resources/qml/RoundedRectangle.qml:24 Expected token `:'

ianrumford, 2019-01-03:
A quick follow up.:Tried to install on a fresh VirtualBox snap and it failed:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 cura : Depends: qml-module-qtquick-controls (>= 5.10) but it is not going to be installed
        Depends: qml-module-qtquick-controls2 (>= 5.10) but it is not going to be installed
        Depends: qml-module-qtquick-dialogs (>= 5.10) but 5.9.5-0ubuntu1 is to be installed
        Recommends: fonts-open-sans but it is not going to be installed
        Recommends: cura-resources-firmware but it is not going to be installed
        Recommends: cura-plugins-all but it is not going to be installed
        Recommends: python3-numpy-stl
        Recommends: python3-pyqt5.qtwebsockets but it is not going to be installed
        Recommends: python3-pyqt5.qtx11extras but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

thopiekar, 2019-01-03:
Found a different PPA in the meantime.
Will try it out soon and if it doesn't work, too, then I will remove Cura for Bionic and advertise people to upgrade to Cosmic.
I guess there is no other realistic solution, I see.

ianrumford, 2019-01-03:
For me upgrading from the Bionic LTS is not an option for my principal system but I can use Cosmic under Virtualbox. Not ideal but good enough.

thopiekar, 2019-01-05:
Tried out to use the "Kubuntu CI Stage" PPA. Using this one would result, that most of the KDE desktop would need to be removed. Upgrading Qt+PyQt on Bionic would need a rebuild of all packages, which depend on it, it seems.
On the other hand, rolling back to 3.6 is not possible. The build was broken before, too. So I see no huge benefit in downgrading to the previous code.
In addition to that, I would need to hack the version number of the package to enforce an update to the older version. Therefore I see too few benefits in investing time in a downgrade.

So please remove the PPA via ppa-purge on Ubuntu Bionic and use Canonicals packages with 3.2 (I think). If you really like to keep using the PPA, please upgrade to Cosmic :wink:

closed 2019-01-05 by thopiekar

Zhigalin, 2019-12-24:
@thopiekar
> I will remove Cura for Bionic and advertise people to upgrade to Cosmic.
> I guess there is no other realistic solution, I see.

Hello and thanks for your work.
I'm almost certain that 4.0 can work in KDE Neon https://neon.kde.org/
It's based on Ubuntu 18.04 but uses Qt 5.13 so it should be fine.
Except that there is no packages in the ppa so I can't test it.
Can you please attach here the packages here so I can test'em and if it works you can add it back to the ppa?

Zhigalin, 2020-01-08:
@thopiekar ?