thopiekar/Cura-packaging #64: Missing Shaders?

JeremyRubin, 2019-03-04:
Cura crashes after slicing when I hit 'preview'. Could there be a missing graphics dependency?

```
2019-03-04 14:03:31,955 - DEBUG - [Thread-22] UM.Backend.Backend._backendLog [101]: [Backend] [WARNING] There are empty layers (or layers with empty extruder plans) in the print! Temperature control and cross layer travel moves might suffer.
2019-03-04 14:03:32,499 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [652]: Slicing took 1.9854435920715332 seconds
2019-03-04 14:03:32,499 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [653]: Number of models per buildplate: {0: 1}
2019-03-04 14:03:32,499 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [667]: See if there is more to slice...
2019-03-04 14:03:32,500 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [171]: Socket state changed to Closing
2019-03-04 14:03:32,500 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [193]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2019-03-04 14:03:32,500 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [173]: Socket state changed to Closed
2019-03-04 14:03:32,500 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [811]: Backend quit with return code 0. Resetting process and socket.
2019-03-04 14:03:35,883 - DEBUG - [MainThread] UM.Controller.setActiveStage [159]: Setting active stage to PreviewStage
2019-03-04 14:03:35,891 - DEBUG - [MainThread] UM.Controller.setActiveView [92]: Setting active view to SimulationView
2019-03-04 14:03:35,892 - INFO - [MainThread] STLReader.STLReader.load_file [46]: Using NumPy-STL to load STL data.
2019-03-04 14:03:35,893 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [444]: Calculating normals took 0.0003216266632080078 seconds
2019-03-04 14:03:35,901 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [401]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0076715946197509766 seconds. 351 input vertices. 351 output vertices.
2019-03-04 14:03:35,902 - DEBUG - [MainThread] STLReader.STLReader._read [97]: Loaded a mesh with 12612 vertices
2019-03-04 14:03:35,903 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/usr/lib/cura/plugins/SimulationView/simulationview_composite.shader]...
```

thopiekar, 2019-03-04:
Hey,

noticed the same for one or two weeks. Always skipped the preview tab, because I hoped it is an error related to the nvidia drivers I have installed or something to my PC.
Are you using Nvidia graphics, too. Don't think it is really the issue, but curious.

JeremyRubin, 2019-03-04:
It may or may not be the issue, but I am indeed using an nvidia card.

JeremyRubin, 2019-03-04:
FYI:

```
>>> import ctypes
>>> from ctypes.util import find_library as f
>>> l = f("GL")
>>> ctypes.CDLL(l, ctypes.RTLD_GLOBAL)
<CDLL 'libGL.so.1', handle 5624d6fdf700 at 7f15c8fa8410>
>>> 
```

Ghostkeeper, 2019-03-05:
Are you using the AppImage release? Or Thopiekar's PPA release?

thopiekar, 2019-05-04:
Can't reproduce the problem with the shaders, but made Cura working again.
Please reopen if you still have the problem.

closed 2019-05-04 by thopiekar