thopiekar/Cura-packaging #65: 18.10 fails to run

biscuitlad, 2019-03-05:
Hi
latest ubuntu and updated after adding the PPA:

xx@localhost:~$ lsb_release -a 
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.10
Release:        18.10
Codename:       cosmic

PPA stable update:
xx@localhost:~$ cura --version
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 17, in <module>
    from PyQt5.QtCore import QT_VERSION_STR, PYQT_VERSION_STR, Qt, QUrl
ImportError: libQt5Core.so.5: cannot open shared object file: No such file or directory

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/cura", line 129, in <module>
    from cura.CuraApplication import CuraApplication
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 11, in <module>
    from PyQt5.QtCore import QObject, QTimer, QUrl, pyqtSignal, pyqtProperty, QEvent, Q_ENUMS
ImportError: libQt5Core.so.5: cannot open shared object file: No such file or directory

Any hints?

thopiekar, 2019-03-05:
Hey @biscuitlad 

You catched an interesting bug.
Normally libQt5Core.so.5 should be provided by the package called libqt5core5a.
That should be ensured by the following chain of depending packages:
cura -> python3-uranium -> python3-pyqt5 -> libqt5core5a
Personally, I see no reason, why this should be not the case on your computer.
All of these packages depend (strictly!) on each other. So if cura is installed libqt5core5a must be there. 

Is there anything special about your installation? Using other PPAs? Tweaked your package manager aka APT?

Thanks!

biscuitlad, 2019-03-05:
Hiya

Yeah, using one other PPA for blender 2.80?

Other than that, it's a straightforward distro upgrade from bionic. 

What should I check?

thopiekar, 2019-03-08:
Nope, that shouldn't be the problem. Check whether libqt5core5a is installed.
It must be there actually, but have no idea how it comes to your error message.

biscuitlad, 2019-03-13:
So,  tried installing those libs:

cosmic)xxxx@localhost:~$ sudo apt install libqt5core5a
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libqt5core5a is already the newest version (5.11.1+dfsg-7ubuntu2).
libqt5core5a set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

(cosmic)xxxx@localhost:~$ sudo apt install python3-uranium
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-uranium is already the newest version (1:4.0~201903040852~rev2641~pkg295~ubuntu18.10.1).
python3-uranium set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

What does the 3 not upgraded mean? Is that possibly the problem?

thopiekar, 2019-03-13:
Don't know. You will see when running `sudo apt dist-upgrade` :wink:

biscuitlad, 2019-03-13:
Yeah, not related. I'm at a loss what to do?

(cosmic)xxxx@localhost:~$ sudo apt dist-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  xserver-xorg-core xserver-xorg-video-dummy xserver-xorg-video-intel
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

biscuitlad, 2019-03-13:
Does this help?

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/cura", line 129, in <module>
    from cura.CuraApplication import CuraApplication
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 11, in <module>
    from PyQt5.QtCore import QObject, QTimer, QUrl, pyqtSignal, pyqtProperty, QEvent, Q_ENUMS
ImportError: libQt5Core.so.5: cannot open shared object file: No such file or directory

(cosmic)xxxx@localhost:~$ whereis libQt5Core.so.5 | xargs  ls -al
ls: cannot access 'libQt5Core.so:': No such file or directory
lrwxrwxrwx. 1 root root 20 Nov  2 11:41 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 -> libQt5Core.so.5.11.1

biscuitlad, 2019-03-13:
OK, I had to downgrade (purge your PPA) and run:

sudo strip --remove-section=.note.ABI-tag /usr/lib/x86_64-linux-gnu/libQt5Core.so.5

and that restored Cura 3.3.1 to a working version. 

Sadly, when I added your PPA and tried it again, it failed with a new error. I guess at this point I have to give up on it!

Ghostkeeper, 2019-03-14:
New error usually means progress! :laughing:

thopiekar, 2019-03-14:
@biscuitlad Personally, I would never touch any system files. When doing this, you never can go back to Ubuntu//Canonical and complain about bugs since the file on your computer is not the same as they delivered in their packages.
Or where do you have this command from?

biscuitlad, 2019-03-14:
I think there may be a problem with the kernal version in the upgraded ubuntu. If you do a search on not being able to share / find that libQt5Core.so.5, you'll come across it periodically and others from Qt. 

Like you say, I did it as I had nothing to lose. I use ubuntu on a chromebook via crouton just so I can run Cura and Blender. If I can't run Cura, then I might as well delete the chroot. Unfortunately there's no support for cosmic currently, so I was forced to do a dist-upgrade. And that's when I hit the libQt5Core nightmare... there are no good fixes out there it would seem.

thopiekar, 2019-05-04:
Ok, that seems to be a customized installation. Hope you managed to get it working.

closed 2019-05-04 by thopiekar

biscuitlad, 2019-05-07:
No I gave up and installed Slic3r PE via fresh install of Debian. I won't
come back to python lib hell...