thopiekar/Cura-packaging #66: Cura 4.0 support in Linux Mint?

ksoviero, 2019-03-12:
You make a point of mentioning that you need Ubuntu 18.10+ to use Cura 4.0, however, all versions of Linux Mint (the most(?) popular distro at the moment, afaik) are based on the Ubuntu LTS releases, i.e., the latest Linux Mint is still based on 18.04, not 18.10.

So, from what I gather, there will be no support for Cura outside of Ubuntu proper?

ksoviero, 2019-03-22:
@thopiekar ?

thopiekar, 2019-03-22:
Oh, sorry.

There is, but that depends whether the current version works well with the latest version of Cura.
For example 4.0.0 is out now and does not work on the latest Ubuntu LTS thus also not on Mint.
Additionally there are still bugs when running 4.0.0 due to some software dependencies on Ubuntu, which differ to what Ultimaker uses.
So when thinking of the problem I had with 3.6 and the issue with 4.0 now, I agree that the PPA is currently not the most optimal way of using Cura at the moment(!).

Personally, I also hope that it will get better as it was years before. But right now I can't do and change much about the situation.

Ghostkeeper, 2019-03-26:
Hopefully with Ubuntu 19.04 LTS in a month we'll get some packages updated for Mint as well.

I just switched my installation to use Qt 5.12 and PyQt 5.12 and that works fine, by the way.

thopiekar, 2019-03-26:
Oh, they are going to offer an LTS that soon? Had something like one LTS per 2 years in mind.

According to what I see on Launchpad's page, they are still using 5.11 at the moment.
https://launchpad.net/ubuntu/+source/pyqt5
Same for Qt5:
https://launchpad.net/ubuntu/+source/qtbase-opensource-src

Personally, I doubt that they are going to upgrade it that late. 😕

Ghostkeeper, 2019-03-28:
Oh, yeah sorry the 19.04 is not LTS of course. That will be only 20.04 so it'll take a while until Mint will have their base package versions updated.

thopiekar, 2019-04-22:
19.04 is not going to be the next LTS, but it has Qt 5.12 included. Was pretty sure that Canonical is not going to make that upgrade that late, but they did!

thopiekar, 2019-04-24:
Still the same with Ubuntu Disco and Qt 5.12.
Even worse: I can't skip the wizard now.
Is there a hack, so I can skip it? @Ghostkeeper

Ghostkeeper, 2019-04-25:
If you were to place a file in $XDG_CONFIG_HOME/cura/cura.cfg then it skips the welcome screen and some miscellaneous welcome pages (sharing data, creating account). If that file were to contain the current version number then it also skips the change log. If that file were to contain a valid container stack and the $XDG_DATA_HOME/cura/machine_instances folder contains such a container stack, then it also skips the adding of a printer.

thopiekar, 2019-04-25:
$XDG_CONFIG_HOME/cura/cura.cfg is already there, but the screen still appears.
To be honest, it goes more on my nerves that there are these display errors. I think this is the main cause and there must be a solution for this. Sure I can hack around, but this won't fix the main issue.

thopiekar, 2019-04-25:
Ok, seems to be KDE related. This is what we see on KDE: (after going through first start wizard)
[image]

And this is what I get in Gnome:
[image]

So it looks like there must be a way to skip KDE's QML middleware. In the logs of the Cura issue, there are warnings visible. I think they should be taken more serious I think.
@fieldOfView: Do you know more about how these KDE widgets get in between? Couldn't find any environment variable, which could tell Qt to use KDE's widgets in between.

fieldOfView, 2019-04-26:
Not my area of expertise. I doubt KDE's QML middleware is used; As far as I know PyQt links fairly directly to Qt (and its implementation of QML).

Ghostkeeper, 2019-04-29:
No, PyQt can use the system's Qt library files, so if KDE has modified their version of Qt then it could use those. Binary compatibility is something the Qt community has paid attention to.

thopiekar, 2019-04-29:
But then it doesn't make sense. On both KDE and Gnome you are using the same Qt libs. I'll ask the KDE community soon, whether there is an plugin or something like that, so they can inject some code into QML.

thopiekar, 2019-05-04:
Ok, short update:

I didn't get any feedback from the KDE community, but while hacking around I noticed that unsetting one of these does the trick:
$ unset KDE_FULL_SESSION 
$ unset KDE_SESSION_UID 
$ unset KDE_SESSION_VERSION

thopiekar, 2019-05-04:
Ok, was a combination together with other unsets...

thopiekar, 2019-05-04:
To summarize: Please upgrade to Ubuntu Disco or Ubuntu Cosmic at least.
I think I managed to fix the problem mentioned above.

closed 2019-05-04 by thopiekar