thopiekar/Cura-packaging #67: Extruder start and end gcodes not working

maukcc, 2019-03-28:
The Extruder start and end gcodes work only for a limited amount of characters

A gcode like:

> ;start T0
> 
> M117 changing tool....
> M109 T0 S{material_print_temperature}
> 
> G1 E6 F120; prime
> G1 E4 F1200
> G92 E0
> G1 Y-100 F3000; wipe
> G1 X-44 F9000
> G1 Y-47
> 
> M117 printing...

ends up like

> P�ș�U

This happens in stable and master.
It does not happen with the appimage or on other OS

thopiekar, 2019-03-28:
Huh.. makes no sense to me. @Ghostkeeper do you have an idea how this could come. The engine should have access to all files. And if not I would expect a crash or empty start and stop gcode or what ever. But this data mess looks like a memory leak or whatever.

Ghostkeeper, 2019-03-29:
Yeah this was a bug in CuraEngine. We fixed it here: https://github.com/Ultimaker/CuraEngine/commit/8abc59d60e8b6fe3e0a71f714786c5f78c1666bb
Should get released with Cura 4.1.

thopiekar, 2019-03-29:
Good news. Thank you for sharing that! 👍

thopiekar, 2019-03-29:
By the way, since the PPA is a fragile thing already. Is it save to use all the software made for 4.0 together with the engine built from master?
Usually I'm not a fan of making mixes like that, but things like the preview tab are broken already, so it shouldn't matter too much. 😅

Ghostkeeper, 2019-03-29:
No, I don't think so. I thought we added some new settings in 4.1 related to the order in which the brim is printed vs. the prime tower (or was it support?). So the new CuraEngine will complain that it needs a value for those settings.

It could work if you also take the fdmprinter.def.json file from master.

thopiekar, 2019-03-29:
Hmm, I better wait then :sweat_smile:
Thank you for the tech. details, @Ghostkeeper !

closed 2019-05-04 by thopiekar