thopiekar/Cura-packaging #71: Latest Master Package Breaking on Ubuntu 19.10

bieber, 2019-12-25:
After the latest update from the master PPA, I now get this in the terminal when attempting to run `cura`:

```
bieber@bieber-laptop:~$ cura
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/usr/bin/cura.real", line 15, in <module>
    import sentry_sdk
ModuleNotFoundError: No module named 'sentry_sdk'
```

This is from version 1:4.4~201912241018~rev4294~pkg328~ubuntu19.10.1

Ghostkeeper, 2019-12-30:
That's a new dependency for the upcoming Cura version so we'd need to bring in that dependency here. It can be installed via Pip.

bieber, 2020-01-01:
I'm assuming that's something that needs to be done in the packaging script, right?  I have it installed in my local env but Cura still crashes, so presumably it has its own import path

Ghostkeeper, 2020-01-01:
I would think that installing it in your local Python3 environment would fix it. I'm no expert in the PPA packaging system though and I don't know what Python version it is using.

Of course the user won't normally need to do anything, ideally.

bieber, 2020-01-08:
@thopiekar, is there a requirements.txt or similar file in here I can patch to add this new dependency?

thopiekar, 2020-01-11:
Nope, requirements for Cura are listed under Cura/control. However, the software mentioned there are packages in Ubuntu and SentrySDK is not in Ubuntu's repos.
Therefore, you need to package SentrySDK for Ubuntu/Debian first.
Anyway, Cura should run without SentrySDK now as my PR has been merged in Ultimaker/Cura.

thopiekar, 2020-01-11:
https://github.com/Ultimaker/Cura/pull/6906

thopiekar, 2020-01-11:
@Ghostkeeper You are right. It can be installed using `pip3`.
But then it will only apply to the users account unless you are using `sudo pip3`, which I wouldn't do to prevent data mess.

closed 2020-01-11 by thopiekar