thopiekar/Cura-packaging #73: Cura from PPA won't install on Focal (20.04)

iggy, 2020-05-02:
When trying to install to Focal (20.04), I'm running into errors that won't let the install proceed.

```
The following packages have unmet dependencies:
 cura : Depends: python3-uranium (>= 4.4.0) but it is not going to be installed
        Depends: python3-uranium (< 4.5) but it is not going to be installed
        Depends: uranium-plugins (< 4.5) but 1:4.4.0~202001031748~rev2946~pkg328~ubuntu20.04.1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

Tracing the deps down manually, I finally get to a, possibly, more useful error:

```
sudo apt install python3-arcus
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-arcus : Depends: python3 (< 3.8) but 3.8.2-0ubuntu2 is to be installed
E: Unable to correct problems, you have held broken packages.
```

Maybe the packages need to be rebuilt against the python version that ended up shipping with Focal?

Steps to reproduce:
```
sudo add-apt-repository ppa:thopiekar/cura
sudo apt install cura
```

Let me know if I can provide any further info

thopiekar, 2020-05-04:
Hey @iggy,

looks like the stable PPA needs an update after some weeks. I'm putting this onto my todo list.

SamSpiri, 2020-06-20:
got the same problem here

Ghostkeeper, 2020-06-30:
I'm running Cura on Focal through Python 3.8, which seems to work fine. I think the package dependencies can be loosened somewhat then.

norro, 2020-07-27:
I am experiencing the same.

thopiekar, 2020-08-28:
Found some time again to look into the PPAs again. I'm rolling out 4.7 now. Could someone retry at the beginning of next week?
The rollout might take some time.

thopiekar, 2020-08-31:
How is the situation gentlemen? :)

iggy, 2020-08-31:
Works for me now. Thank you ☺️

norro, 2020-08-31:
Works for me. Thank you :)

thopiekar, 2020-09-01:
Great! Guess we can close this one then.
Thank you for keeping patient! :+1:

closed 2020-09-01 by thopiekar