thopiekar/Cura-packaging #75: cura-master (4.8) PPA needs pynest2d

iggy, 2020-11-18:
Apparently https://github.com/Ultimaker/pynest2d is now a dependency for 4.8.

Dinth, 2020-12-26:
also experiencing this issue

```
Traceback (most recent call last):
  File "/usr/bin/cura.real", line 25, in <module>
    import pynest2d # @UnusedImport
ModuleNotFoundError: No module named 'pynest2d'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 153, in apport_excepthook
    with os.fdopen(os.open(pr_filename,
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_bin_cura.real.1000.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/cura.real", line 25, in <module>
    import pynest2d # @UnusedImport
ModuleNotFoundError: No module named 'pynest2d'

```

thopiekar, 2021-03-27:
Added

closed 2021-03-27 by thopiekar