thopiekar/Cura-packaging #81: Update to 5.0

dxgldotorg, 2022-05-22:
Cura 5.0 recently came out, with updated Qt dependencies and a new version of the plugin SDK.

In addition, please upload 5.0 compatible versions of your plugins to the Cura Marketplace, as all your plugins have been removed due to being incompatible with 5.0/Qt6.

LightTemplar, 2022-05-27:
And I also ask about packages for Ubuntu 22.04.

gamersi, 2022-06-12:
@thopiekar We need 5.0 support.

thopiekar, 2022-06-18:
Got something for you to test :slightly_smiling_face: 
Basically follow these instructions and install `com.ultimaker.cura` instead.

thopiekar, 2022-06-18:
PS: I'll do some changes soon, like shrinking the package. There are some development files in there still, but at least it runs for me (AMD and Intel).

iggy, 2022-09-14:
Maybe a note in the repo/your website to make it easier to find out you've discontinued this? I think I looked at 10 separate pages before finally stumbling on this (closed) issue that says it's discontinued.

LightTemplar, 2022-09-15:
@iggy, like this? 
[image]

iggy, 2022-09-15:
Yeah. Something exactly like that. Except in one of the top 5 Google results for "cura ppa". Or even a page linked from one of the top 5 results.